chirpstack/chirpstack-gateway-os #1: bitbake lora-gateway-os-base error

jprodriguez-uutech, 2019-01-15:
lora-packet-forwarder_4.0.1.bb fails with the following errors:

yocto@37458b161cfb:/build$ **bitbake lora-gateway-os-base**
Loading cache: 100% |##################################################################################################################################################################| Time: 0:00:01
Loaded 3089 entries from dependency cache.
WARNING: /lora-gateway-os/layers/loraserver/meta-semtech/recipes-lora/lora-packet-forwarder/lora-packet-forwarder_4.0.1.bb: Unable to get checksum for lora-packet-forwarder SRC_URI entry global_conf.eu868.json: file could not be found
WARNING: /lora-gateway-os/layers/loraserver/meta-semtech/recipes-lora/lora-packet-forwarder/lora-packet-forwarder_4.0.1.bb: Unable to get checksum for lora-packet-forwarder SRC_URI entry global_conf.eu868.json.gps: file could not be found
WARNING: /lora-gateway-os/layers/loraserver/meta-semtech/recipes-lora/lora-packet-forwarder/lora-packet-forwarder_4.0.1.bb: Unable to get checksum for lora-packet-forwarder SRC_URI entry global_conf.us915.json: file could not be found
WARNING: /lora-gateway-os/layers/loraserver/meta-semtech/recipes-lora/lora-packet-forwarder/lora-packet-forwarder_4.0.1.bb: Unable to get checksum for lora-packet-forwarder SRC_URI entry global_conf.us915.json.gps: file could not be found
Parsing recipes: 100% |################################################################################################################################################################| Time: 0:02:06
Parsing of 2131 .bb files complete (0 cached, 2131 parsed). 3089 targets, 149 skipped, 0 masked, 0 errors.
ERROR: No recipes available for:
  /lora-gateway-os/meta/recipes-mender/mender/mender_%.bbappend

Summary: There were 4 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

brocaar, 2019-01-16:
> ERROR: No recipes available for:
/lora-gateway-os/meta/recipes-mender/mender/mender_%.bbappend

It looks like your `/build/conf/bblayers.conf` file is missing the mender layers:
https://github.com/brocaar/lora-gateway-os/blob/master/meta/conf/bblayers.conf.sample

> WARNING: /lora-gateway-os/layers/loraserver/meta-semtech/recipes-lora/lora-packet-forwarder/lora-packet-forwarder_4.0.1.bb: Unable to get checksum for lora-packet-forwarder SRC_URI entry global_conf.us915.json.gps: file could not be found

A `git pull` should solve this issue. There were a bunch of typos (RAK381 vs RAK831).

jprodriguez-uutech, 2019-01-16:
Thanks, I will try it

 

https://github.com/brocaar/lora-gateway-os/blob/master/meta/conf/bblayers.conf.sample

jprodriguez-uutech, 2019-01-17:
Hello Orne,

 

I have removed the old version and cloned the repository again. It is still complaining about the mender recipe:

 

jprodriguez@u18-lora:~/git$ git clone https://github.com/brocaar/lora-gateway-os.git

Cloning into 'lora-gateway-os'...

remote: Enumerating objects: 591, done.

remote: Counting objects: 100% (591/591), done.

remote: Compressing objects: 100% (369/369), done.

remote: Total 591 (delta 183), reused 559 (delta 161), pack-reused 0

Receiving objects: 100% (591/591), 230.48 KiB | 904.00 KiB/s, done.

Resolving deltas: 100% (183/183), done.

jprodriguez@u18-lora:~/git$ cd lora

-bash: cd: lora: No such file or directory

jprodriguez@u18-lora:~/git$ cd lora-gateway-os/

jprodriguez@u18-lora:~/git/lora-gateway-os$ sudo su

[sudo] password for jprodriguez:

root@u18-lora:/home/jprodriguez/git/lora-gateway-os# make submodules

git submodule init

Submodule 'bitbake' (git://git.openembedded.org/bitbake) registered for path 'bitbake'

Submodule 'layers/bsp/meta-raspberrypi' (git://git.yoctoproject.org/meta-raspberrypi) registered for path 'layers/bsp/meta-raspberrypi'

Submodule 'layers/mender/meta-mender' (https://github.com/mendersoftware/meta-mender.git) registered for path 'layers/mender/meta-mender'

Submodule 'layers/openembedded/meta-openembedded' (git://git.openembedded.org/meta-openembedded) registered for path 'layers/openembedded/meta-openembedded'

Submodule 'layers/openembedded/openembedded-core' (git://git.openembedded.org/openembedded-core) registered for path 'layers/openembedded/openembedded-core'

git submodule update

Cloning into '/home/jprodriguez/git/lora-gateway-os/bitbake'...

Cloning into '/home/jprodriguez/git/lora-gateway-os/layers/bsp/meta-raspberrypi'...

Cloning into '/home/jprodriguez/git/lora-gateway-os/layers/mender/meta-mender'...

Cloning into '/home/jprodriguez/git/lora-gateway-os/layers/openembedded/meta-openembedded'...

Cloning into '/home/jprodriguez/git/lora-gateway-os/layers/openembedded/openembedded-core'...

Submodule path 'bitbake': checked out '0cfc71d1a342b82781b0ba547421e41f6340902a'

Submodule path 'layers/bsp/meta-raspberrypi': checked out 'fa217ed21cedc58e13dda23a0237134333787a01'

Submodule path 'layers/mender/meta-mender': checked out 'c49d2d0f9bf073bf9ff3898154791869c9a922ee'

Submodule path 'layers/openembedded/meta-openembedded': checked out 'a15d7f6ebcb0ed76c83c28f854d55e3f9d5b3677'

Submodule path 'layers/openembedded/openembedded-core': checked out '0ac1b496823a57ac6a234a14c60318bf1ffa19b0'

root@u18-lora:/home/jprodriguez/git/lora-gateway-os# docker-compose run --rm yocto bash

yocto@b77924a988be:/lora-gateway-os$ source oe-init-build-env /build/ /lora-gateway-os/bitbake/

yocto@b77924a988be:/build$ bitbake lora-gateway-os-base

Loading cache: 100% |##################################################################################################################################################################| Time: 0:00:01

Loaded 3089 entries from dependency cache.

Parsing recipes: 100% |################################################################################################################################################################| Time: 0:02:04

Parsing of 2131 .bb files complete (0 cached, 2131 parsed). 3089 targets, 149 skipped, 0 masked, 0 errors.

ERROR: No recipes available for:

  /lora-gateway-os/meta/recipes-mender/mender/mender_%.bbappend

 

Thanks,

Juan Pedro

 

https://github.com/brocaar/lora-gateway-os/blob/master/meta/conf/bblayers.conf.sample

brocaar, 2019-01-23:
Did you call `make submodules`?

jprodriguez-uutech, 2019-01-24:
Yes, but outside the container. Inside the container does not run.

closed 2019-01-24 by brocaar

jprodriguez-uutech, 2019-01-24:
> I'm now testing this in a Linux environment (was testing this on MacOS) and there are some differences in how file permissions are handled (as with MacOS the volume share works different).
> I'm running in some permission errors and will update the README.md instructions to solve these :-)

Thanks

commit 60d746f1 (2019-01-24): Update documentation to deal with Docker permission issues.

Closes #1.