chirpstack/chirpstack-gateway-os #4: iC980a (915 MHz Version of iC880a) no 915 MHz Config available

fvi21, 2019-03-03:
Hi,
excellent work by the way, at this point I've tested the most simple things of the raspberry pi image, but with some problems at this point.
In the latest version of the code I recently checked there's no longer support for the 915 MHz of the packet forwarder. When I try to setup the gateway in the raspberry pi with the iC980a (the 915MHz version of the iC880a) it doesn't start the packet-forwarder, I checked the reset pin AND the global_config.json checking that there's everything fine in the config file. Nothing good after all because the monit status always response with a failure.
I downloaded and flashed the older version, the December 2018 version, and it has the 915 MHz profile and it worked perfectly. 
Maybe I'm wrong and wasting your time or maybe it's something to improve, I appreciate your time and hope you can answer if there's any kind of new way to config the iC980a in the newest OS version or it's not supported at all and have to find another solution.
Thanks in advance.

brocaar, 2019-03-04:
Is the iC980a still being sold? I could not find it at https://www.wireless-solutions.de/products/long-range-radio.html.

I removed it from the iC880a as this model is for EU868 only, so it didn't make sense to have an US915 channel-plan listed here.

If you would like to contribute to add an iC980c option, that would be great. See also https://github.com/brocaar/lora-gateway-os/blob/master/layers/targets/meta-raspberrypi/recipes-core/gateway-config/gateway-config/gateway-config.sh#L14.

fvi21, 2019-03-04:
Yes, of course, I'm working to make this compile and work with your OS. But I have to say I'm new with GITHUB and I don't know yet how to send a change or something in the code. I'll update this post and put everything I change in the configurations to add it as I made it work.
About the iC980a, it's not listed as a default option to buy. I requested an special one to make a LoRaWAN concentrator under the South America Standard and I'm developing one so probably I'll be in touch in the forum requesting help with the "front-end" part. But first I must have to make this work correctly to have no more need to touch the gateway.
Thanks for the answer, and I'll post here the changes and I'll read here or probably google how to work with github. Thanks, and greetings from Argentina.

fvi21, 2019-03-09:
Helllo again,
I'm posting an update after a few days of testings and failures.
To be consistent with your default setup about the configurations, I would like to share the configuration of the global_conf.json wich is running in my concentrator.

[global_conf_US](https://github.com/brocaar/lora-gateway-os/files/2948742/global_conf.txt)

probably you would like to correct the errors in the configuration. I were trying to follow the specifications declared here [LoRaWAN - Regional Parameters - Section 2.3.2 - Page 21](https://lora-alliance.org/sites/default/files/2018-04/lorawantm_regional_parameters_v1.1rb_-_final.pdf)

In the **loraserver.toml** I follow the configuration of the guide posted in your website, so in the channels configuration I have
`enabled_uplink_channels=[0, 1, 2, 3, 4, 5, 6, 7]`
Finally inside the web interface, when I create the Gateway-Profile I put channels
[image]

and inside the service profile the significant config is set the Data Rates, in my case 

```
min DR: 0 
max DR: 3
```

with this config I were able to connect OTAA devices and works perfectly. 

My configuration with the Raspberry Pi 3 and the iC980a SPI version is like this:
```
iC980a PIN 21 (5VCC)	->	Raspi3 PIN 4  (DC POWER 5V)
iC980a PIN 22 (GND)	->	Raspi3 PIN 6  (GND)
iC980a PIN 16 (MOSI)	->	Raspi3 PIN 19 (SPI_MOSI)
iC980a PIN 15 (MISO)	->	Raspi3 PIN 21 (SPI_MISO)
iC980a PIN 13 (RST)	->	Raspi3 PIN 22 (GPIO25)
iC980a PIN 14 (SPI_CLK)	->	Raspi3 PIN 23 (SPI_CLK)
iC980a PIN 17 (NSS)	->	Raspi3 PIN 24 (SPI_CE0)
```

[image]

the extra board there is a GPRS-GPS module. I use it in case I have a downtime with my ISP (it's normal in Argentina) and I modify the scripts to begin the needed AT commands to send me the GPS coordinates in NMEA every 1 second to the serial port which is configured in the global_conf file uploaded here.
Until now, it works fine, now I'm reading how to do an Influx/Grafana integration because I don't know too much about front-ends.

About the contribution in the menu file I have some ideas and here I'll post to ask for your help. I saw the bash code and look at the files in your recipes and I realize it's possible to do something like the iC880A but merged with the RAK831 configuration. I'm trying to take part of the code of the block from the iC880A and from the RAK831 block. So, I have this:
[iC980A.zip](https://github.com/brocaar/lora-gateway-os/files/2948825/iC980A.zip)
The folder like the iC880A, with the Channel Plans like the RAK831 modified like the first one in this post trying to complain with the LoRaWAN Regional Parameters like I said early. Probably you would like to check if I missed in a configuration in the channels plans, particularly in the 6 and 7. And the bash code for the gateway-config.sh I think this is a way to do it:
[gateway-config.txt](https://github.com/brocaar/lora-gateway-os/files/2948834/gateway-config.txt)

I've change the `do_setup_concentrator_shield()` like this:

```
do_setup_concentrator_shield() {
    FUN=$(dialog --title "Setup LoRa concentrator shield" --menu "Select shield:" 15 60 4 \
        1 "IMST     - iC880A" \
	2 "IMST     - iC980A" \
        3 "RAK      - RAK831" \
	4 "RisingHF - RHF0M301" \
        5 "Sandbox  - LoRaGo PORT" \
        3>&1 1>&2 2>&3)
    RET=$?
    if [ $RET -eq 0 ]; then
        case "$FUN" in
            1) do_prompt_concentrator_reset_pin && do_setup_ic880a;;
            2) do_prompt_concentrator_reset_pin && do_setup_ic980a;;
	    3) do_set_concentrator_reset_pin 17 && do_setup_rak831;;
	    4) do_set_concentrator_reset_pin 7  && do_setup_rhf0m301;;
            5) do_set_concentrator_reset_pin 25 && do_setup_lorago_port;;
        esac
    fi
}
```
As you can see there's `2) do_prompt_concentrator_reset_pin && do_setup_ic980a;;` like the iC880A and it goes to:
```
do_setup_ic980a() {
    FUN=$(dialog --title "Channel-plan configuration" --menu "Select the channel-plan:" 15 60 3 \
        1 "US915" \
        3>&1 1>&2 2>&3)
    RET=$?
    if [ $RET -eq 1 ]; then
        do_main_menu
    elif [ $RET -eq 0 ]; then
        case "$FUN" in
            1) do_select_us915_block "iC980A";;
        esac
    fi
}
```
Like the RAK831, but in this case I only left 1 option, in my case US plan option. If I have more time after finish this project I'll probably add here the AU plan. The rest of the code is exactly like you do in your script, but you must add the iC980A folder to the `lora-gateway-os/layers/targets/meta-raspberrypi/recipes-lora/lora-packet-forwarder/lora-packet-forwarder/`. Please try to check it and feel free to modify anything and use it to your project.
Thanks for your time and hope you enjoy the idea. Any suggestions are welcome.

closed 2019-03-15 by fvi21

Luckyman90, 2023-02-09:
Hi fiv21,
It's so good. I am a newbie.  so I wonder if  I Can have a detailed document that guides me on how to configure the IC and Raspi to work.