chirpstack/chirpstack-gateway-os #8: No IMAGE_CMD defined

jprodriguez-uutech, 2019-04-23:
In a fresh-cloned docker, bitbake is failing because of missing configuration.

~/git/lora-gateway-os$ docker-compose run --rm yocto bash
yocto@f7c1d562c3d4:/lora-gateway-os$ source oe-init-build-env /build/ /lora-gateway-os/bitbake/
yocto@f7c1d562c3d4:/build$ bitbake lora-gateway-os-base
ERROR: /lora-gateway-os/layers/bsp/meta-raspberrypi/recipes-core/images/rpi-test-image.bb: No IMAGE_CMD defined for IMAGE_FSTYPES entry 'sdimg' - possibly invalid type name or missing support class ETA:  0:01:47
ERROR: /lora-gateway-os/layers/bsp/meta-raspberrypi/recipes-core/images/rpi-hwup-image.bb: No IMAGE_CMD defined for IMAGE_FSTYPES entry 'sdimg' - possibly invalid type name or missing support class
ERROR: Failed to parse recipe: /lora-gateway-os/layers/bsp/meta-raspberrypi/recipes-core/images/rpi-test-image.bb
ERROR: /lora-gateway-os/layers/bsp/meta-raspberrypi/recipes-core/images/rpi-basic-image.bb: No IMAGE_CMD defined for IMAGE_FSTYPES entry 'sdimg' - possibly invalid type name or missing support class

Summary: There were 4 ERROR messages shown, returning a non-zero exit code.
yocto@f7c1d562c3d4:/build$

jprodriguez-uutech, 2019-04-23:
Possible solution:
add to conf/local.conf the following configuration:
IMAGE_FSTYPES = "tar.xz ext3 rpi-sdimg"

(https://yocto.yoctoproject.narkive.com/N1e5LkgV/how-do-i-used-yocto-to-build-an-sdcard-image-for-my-raspberry-pi-3-model-b)

brocaar, 2019-05-01:
I can't reproduce this issue. Maybe your Git submodules are not up-to-date?

> IMAGE_FSTYPES = "tar.xz ext3 rpi-sdimg"

This is not the solution as Mender is used to generate the SD Card images, it should not use the `rpi-sdimg`.

I suggest you start from a clean environment (Also Docker containers) and test again.

closed 2019-08-12 by brocaar