chirpstack/chirpstack-gateway-os #9: Image does not boot on RPI without HDMI attached

jeremyherbert, 2019-04-29:
Hi,

I loaded this image onto the raspberry pi, but it doesn't seem to boot to ssh unless I have HDMI (or a HDMI dummy) attached to the PI.

There is also a reference to this on the forums: https://forum.loraserver.io/t/lora-gateway-os-not-booting-with-out-hdmi-or-keyboard/4126/3

brocaar, 2019-05-08:
I can not reproduce this issue. My Pi boots fine without HDMI (which is how I'm using this most of the time).

cls02, 2019-05-10:
I have the same issue. Updating the Ethernet configuration as proposed in the forum post did not fix this issue for me.

brocaar, 2019-05-10:
Maybe this is specific to a Raspberry Pi version that I have not tested against? Could you share the steps to reproduce this issue + Raspberry Pi version used?

>  but it doesn't seem to boot to ssh unless I have HDMI

Could you clarify this? Does this mean that the Raspberry Pi **did** get an IP address, but does not accept SSH connections?

jeremyherbert, 2019-05-10:
I just burned the image onto a microsd and booted it up. Using raspberry pi 3B+

No, device does not get an IP.

cls02, 2019-05-10:
Confirm the above, also Raspberry Pi 3+

jeremyherbert, 2019-05-10:
Also, the RAK831 board is on top

azhaganandhan, 2019-05-16:
I'm also facing the same issue ethernet is not enabled but I have not checked with HDMI.
My Raspberry Configuration
 Raspberry Pi 3B+, RAK-831 developer kit
The same issue had been posted by me in Lora server forum
https://forum.loraserver.io/t/how-to-configure-gateway-using-base-os-image/4578

martinoorlandi, 2019-07-02:
Same issue with a RPi 3+ and a RAK2245 hat...
Anyone has found a solution yet?

brocaar, 2019-07-02:
I bought a Pi 3B+ for testing and can not reproduce the issue. My Pi is **only** connected to ethernet + original Raspberry Pi power adapter. The SD card is flashed with https://artifacts.loraserver.io/downloads/lora-gateway-os/raspberrypi/raspberrypi3/2.0.0test4/lora-gateway-os-base-raspberrypi3--20190423145941.sdimg.gz without **any** modifications made.

Afterwards I run (on my laptop which is configured with a network bridge on 10.42.1.1):

```
sudo nmap -sP 10.42.1.1/24
```

```
Starting Nmap 7.70 ( https://nmap.org ) at 2019-07-02 17:22 CEST
Nmap scan report for 10.42.1.252
Host is up (0.00052s latency).
MAC Address: B8:27:EB:1B:7D:BF (Raspberry Pi Foundation)
Nmap scan report for 10.42.1.1
Host is up.
Nmap done: 256 IP addresses (2 hosts up) scanned in 9.60 seconds
```

Then I SSH to the returned IP `admin@10.42.1.252` and I can login.

martinoorlandi, 2019-07-02:
Thanks for your quick reply!

I've tried following your same steps, using Nmap and disconnecting everything but the power and ethernet cable. (the only difference is that I've connected the Pi to a router so that's on the same network with my MacBook).

I've tried also all the combinations of the following:
- powering via my laptop and power adapter 
- with and without the RAK2245 hat 
- lora-gateway-base and lora-gateway-full images (both test4).

Still having the same issue! HDMI cable in and everything works fine, cable out and no ip!

This is head-scratcher!

brocaar, 2019-07-02:
Could this be an issue related to power? The power-adapter seems to be the only difference that we have at this point. Is there a possibility that the HDMI plugged in has influence on the power-usage or is backfeeding power into the Pi (I'm not sure if this is even possible)?

Have you tried without the concentrator shield (to reduce the power usage)?

martinoorlandi, 2019-07-02:
That’s a possibility.

I’ve tried without the concentrator but made no difference.

I’ve tried with a power adapter that is 5V - 2.1A what about yours?

fabioetec, 2019-07-02:
I have the same problem, Raspberry 3B+, both with Ethernet or Wifi.
ic880A Concentrator but I don't think makes a difference.
I think could be relate to the order of boot of something because in some rare cases it works!
Power adapter 5V - 2.5A

martinoorlandi, 2019-07-02:
@fabioetec that’s interesting. Also in my case a couple of times it randomly worked. 

A couple of things that may help narrow down the issue:

- Does this issue only happens with the 3B+. Has anyone experienced it with an earlier version of the RPi 3?
- Does using a HDMI dummy solve the issue? Has anyone tried with consistent good results? In this case we could rule out the HDMI backfeeding power into the RPi suggested by @brocaar

jeremyherbert, 2019-07-02:
I have used a hdmi dummy and it solves the problem. 

Is the raspberry pi firmware on the image the latest (ie the rpi foundation binary driver blobs)?

brocaar, 2019-07-03:
Actually, this problem doesn't seem to be limited to LoRa Gateway OS: https://www.google.com/search?q=raspberry+pi+not+booting+without+hdmi.

What I will do at my end is update the `meta-raspberrypi` to the latest revision (thus latest bootstrap binary, u-boot and kernel) and create a new image. I was planning to do that anyway to create a release containing the v3 components.

martinoorlandi, 2019-07-03:
Thanks for that @brocaar! That seems to explain it and I hope the updated kernel may do the trick.

I’m happy to give it a test when it’s ready.

cls02, 2019-07-05:
Ditto - great that this pesky problem might go away!

Casper

brocaar, 2019-07-07:
Could one of you test this image: https://www.dropbox.com/sh/yofdcrwp6jqk4vc/AABN6IJWtb2-_eqPjITy7Vhba?dl=0&lst=  ?

This:

* Updates all git modules (including the Raspberry Pi) to the latest versions
* Also updates all LoRa Server components to v3

(this is a build of https://github.com/brocaar/lora-gateway-os/tree/v3).

martinoorlandi, 2019-07-08:
@brocaar Thanks for the new build. The ethernet issue seem to be solved for me! 

I've tested on the same environment: RPi 3B+ with a RAK2245 concentrator, 5V 2.1A power adapter and ethernet cable plugged in (no HDMI or anything else).
I've rebooted several times, tried to keep it off for a while and then back on...

Every time it boots up and connects to the network gets its IP fine, no issue at all.

I will keep an eye on it to see if it happens again, but it seem quite promising that I have not managed to reproduce the issue with the new build!

Thanks!

brocaar, 2019-07-08:
That is great news @martinoorlandi  and thanks for testing!

brocaar, 2019-07-15:
This has been released: https://forum.loraserver.io/t/release-lora-gateway-os-v3-0-0test1/5299.

fabioetec, 2019-08-04:
are you sure that it works??
for me doesn't work at all!!!

cls02, 2019-08-05:
Hi,

Works for me.

Casper

splaw1k, 2019-10-15:
Hello,
I still have the problem, that mentioned above. 
My platform: Raspberry Pi 3 + RAK2245 ver. B
I checked it out for the Gateway OS version 3.0 test1 (and test2 too).

brocaar, 2019-11-13:
I'm finally able to reproduce the issue. It seems it is not only related to the HDMI, but also the speed of the SD card. I was only able to reproduce this using a faster / more expensive SD card. See also the above issue I have created.

brocaar, 2019-11-24:
I think I have found the issue. I believe the boot fails because of the initramfs in some cases. When forcing the HDMI output, this error is unfortunately not visible. In most cases all is fine, however depending the speed of the sdcard, the block-device might not be available yet and the initramfs fails to mount the rootfs.

Somehow, connecting the HDMI affects timing of the boot.

closed 2019-11-24 by brocaar

splaw1k, 2019-11-28:
So, do You have some idea to solve this issue (so that it would be works regardless of the speed of SD card) ?

brocaar, 2019-11-28:
Yes, see the above commit. This will be included in the next release.

rogerdahl, 2020-01-28:
Try another SD card. An older 8 GB SD card that works fine with Raspbian did not work for me with the ChirpStack full image. A newer 32 GB card works. This was with
```
chirpstack-gateway-os-full-raspberrypi3-20200127144146.rootfs.wic.gz
```

andershedberg, 2020-09-01:
I see this is fixed so I will need to update, but as I could not find it when googling and I get it with the HDMI cable attached sometimes, I write a short note here if someone else needs to see it. 

I use a Raspberry 3B+ with a Sandisk card detected "new high-speed SDIO card at address 0001" is detected. The error message is:

mount: mounting /dev/mmcblk0 on /mnt/root/ failed: no such file or directory

This happens about 1 time out of 4 reboots with a HDMI cable attached.

brocaar, 2020-09-01:
@andershedberg actually, the HDMI is not the reason why it was failing. It was because of a race-condition in the boot process (maybe a HDMI cable changes the timing, but it is not the real fix). This was the real fix:

https://github.com/brocaar/chirpstack-gateway-os/blob/master/meta/recipes-filesystem/initramfs-readonly-rootfs-overlay/files/init-readonly-rootfs-overlay-boot.sh#L69

andershedberg, 2020-09-01:
yes, my earlier solution to this was to change the display settings in the boot-configuration like in this thread https://www.raspberrypi.org/forums/viewtopic.php?t=11259 It helped, but not always. Glad it was something as simple as a 1 second sleep and the root cause was found! as the gateways caused me sleepless nights and got me into running Mikrotiks gateways this year ;-)
As a side-note - I have got reports about "commercial" gateways that gets stuck in similar ways. Might be a common problem for embedded linux-systems.

commit 5cc92d41 (2019-07-15): Update to v3 components + update openembedded layers.

Closes #9.
commit 48b42516 (2019-11-24): Wait for device before mount_and_boot.

Depending the sdcard speed / brand / ... and if the HDMI is connected it is
possible that device is not yet initialized (/dev/...) at the time the initramfs
script is started and because of this mounting the rootfs fails.

Closes #9.