chirpstack/chirpstack-gateway-os #10: Getting gateway added to vendor list?

shawaj, 2019-05-04:
Hi there,

What's the process to get our gateway added here - https://www.loraserver.io/lora-gateway-os/overview/

Thanks,
Aaron

brocaar, 2019-05-06:
That depends on the platform. E.g. when it is a completely new platform, we would need to add a new BSP layer (`layers/bsp`) and prepare the Mender patches for the bootloader (found under `layers/targets`).

But I assume you're talking about adding support for: https://uk.pi-supply.com/products/iot-lora-gateway-hat-for-raspberry-pi?

In that case, the Raspberry Pi platform is already supported and this would require:

- [ ] Add concentrator shield: https://github.com/brocaar/lora-gateway-os/blob/master/layers/targets/meta-raspberrypi/recipes-core/gateway-config/gateway-config/gateway-config.sh
- [ ] Include concentrator specific config: https://github.com/brocaar/lora-gateway-os/blob/master/layers/targets/meta-raspberrypi/recipes-lora/lora-packet-forwarder/lora-packet-forwarder_%25.bbappend
- [ ] Potentially include packet-forwarder modifications?

shawaj, 2019-05-07:
@brocaar yes that is the Gateway HAT we are talking about.

CC @ryanteck - can you sort the other bits?

ryanteck, 2019-05-07:
Hi @brocaar ,

Yes it is Raspberry Pi based, as it uses the RAK833 which is the same semtech chipset as the ic880a, RAK831 etc.

I've tested it out by selecting ic880a and it worked fine. The GPIO pin used is BCM22 for reset.

We'd be looking for it to have two regions that they can select from either EU868 or US915.

Would you like me to try and figure out what parts to change and do a git pull request?

- Ryan

brocaar, 2019-05-08:
@ryanteck that would be great :-)

ryanteck, 2019-05-17:
Hi @brocaar ,

I believe I've done the changes however haven't been able to test yet. If you can just look through to see if I have done it correctly.

Thanks

brocaar, 2019-05-17:
Thanks @ryanteck :+1: I've provided some feedback in the PR (tiny changes).

brocaar, 2019-07-02:
@ryanteck @shawaj anything I can help with to get your pull-request (https://github.com/brocaar/lora-gateway-os/pull/13) merged? It hasn't been updated for a while. @ryanteck are you still working on this?

ryanteck, 2019-07-03:
Hi @brocaar, was the new PR all fine?

brocaar, 2019-08-12:
Closing this issue as the PR has already been merged :slightly_smiling_face:

closed 2019-08-12 by brocaar