chirpstack/chirpstack-gateway-os #11: no ethernet with test4-image

toepi, 2019-05-05:
want try out mender upgrade and so I flashed test2 to my raspberrypi3 b+. After upgrade it to test4 via a mender (URL: https://artifacts.loraserver.io/downloads/lora-gateway-os/raspberrypi/raspberrypi3/2.0.0test4/) and a reboot it lost the ethernet connection, back to test2 ethernet was there again.

brocaar, 2019-05-14:
I'm trying to reproduce your issue as I have not seen this before.

brocaar, 2019-05-14:
I've:

- [x] Flashed a Raspberry Pi 3 with the base test2 image
- [x] Scanned with `nmap` which IP was assigned (wired) using DHCP
- [x] Used mender to update to the base test4 image
- [x] Successfully logged in again using SSH using the IP of the Raspberry Pi

I don't see any issue with regards to lost ethernet connection.

toepi, 2019-05-14:
strange, same workflow her. okay use a local console instead of nmap for determine assigned ip.

have extra switch back to test2 to check if I've destroy the configuration. and after it flash again to test4 ... with no success. will try it again and will report it her again.

brocaar, 2019-05-21:
I will order a RPi 3 B+ to see if there is any difference between B and B+.

martinoorlandi, 2019-07-02:
Any success anyone?

I have the same issue of no ethernet connection with both images test2 and test4 using a RPi 3B+ with a RAK2245 hat.

It works fine if connected to a HMDI monitor, if I try to boot without it, it doesn't seem to work...not sure if it does not boot at all or just doesn't get an IP address assigned.

brocaar, 2019-07-02:
I still can not reproduce the issue with a Raspberry Pi 3B+. See #9 .

martinoorlandi, 2019-07-02:
Thanks! this Is what I tried: (https://github.com/brocaar/lora-gateway-os/issues/9#issuecomment-507756921)

brocaar, 2019-07-15:
This should be fixed by https://github.com/brocaar/lora-gateway-os/issues/9.

closed 2019-07-15 by brocaar