chirpstack/chirpstack-gateway-os #12: Can't connect

x893, 2019-05-17:
Hi,

- i download lora-gateway-os-full-raspberrypi3--20190423151801.sdimg.gz

- Write it to SD card 8GB

- Connect RPi 3 Model B Ver 1.2 via Ethernet cable to computer (192.168.1.2) and serial line (pin 15,16) to USB-Serial adapter. No RAK831 connected to RPi.

- Add console=serial0,115200 to cmdline.txt

- Insert SD to RPi.

- No replay (ping/ssh) to 192.168.1.5

- No login prompt on serial console

- If i add init=/bash/sh than i see prompt on serial console and can see filesystem

Can you help or irection to resolve ?

Thanks in advance

[putty.zip](https://github.com/brocaar/lora-gateway-os/files/3191417/putty.zip)

brocaar, 2019-05-21:
The Raspberry Pi gets its IP from a DHCP server. Does your computer provide one for the connection to the Raspberry Pi?

x893, 2019-05-21:
Thanks, i try connect via hub.

x893, 2019-05-23:
Thanks ! Connect to router via cable. Set wireless parameters and now works fine.

closed 2019-05-23 by x893