chirpstack/chirpstack-gateway-os #17: Support for Raspberry pi 4

velegip, 2019-08-11:
Hi!

The current version is 3.0.0-test2 is support up to pi3.
Can you modify the image for the new raspberry pi 4?

Thank you!

brocaar, 2019-08-12:
Unfortunately this is not (yet) possible as the Thud version of the meta-raspberrypi layer does not provide this machine (yet?): http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/tree/conf/machine?h=thud.

It is available in the Warrior version, but this would require to update the other layers: http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/tree/conf/machine?h=warrior.

velegip, 2019-08-12:
I see. Thank you.

Then I'll wait for it.

hotmonitor, 2019-12-06:
Hello 
 I test IMG chirpstack-gateway-os-full-raspberrypi3--20191102145611.sdimg in Raspberry Pi4 Buster but img not work. Do you plan to put Pi4 img version and when? Thanks

brocaar, 2020-03-27:
Yes I do plan to add support for the Pi 4. I just received one for testing.

brocaar, 2020-03-28:
Please see #45

commit 2550bc37 (2020-03-28): Add support for Raspberry Pi 4.

Closes #17.
commit 7b71722f (2020-04-02): Add support for Raspberry Pi 4.

Closes #17 and #45.
closed 2020-04-02 by brocaar