chirpstack/chirpstack-gateway-os #18: Image will not build via Yocto because git.drogon.com is no longer available

proffalken, 2019-08-13:
When building a new image from Yocto for deployment via Mender, it currently fails with the following error:

```

ERROR: wiringpi-git-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/lora-gateway-os/layers/openembedded/openembedded-core/scripts:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot/usr/bin/crossscripts:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/usr/sbin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/usr/bin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/sbin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/recipe-sysroot-native/bin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/lora-gateway-os/bitbake/bin:/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/hosttools"; export HOME="/media/yocto/jenkins"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror git://git.drogon.net/wiringPi /media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/downloads/git2/git.drogon.net.wiringPi --progress failed with exit code 128, output:

Cloning into bare repository '/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/downloads/git2/git.drogon.net.wiringPi'...

fatal: unable to connect to git.drogon.net:

git.drogon.net[0: 188.246.205.22]: errno=Connection refused

git.drogon.net[1: 2a03:9800:10:7b::2]: errno=Network is unreachable

ERROR: wiringpi-git-r0 do_fetch: Fetcher failure for URL: 'git://git.drogon.net/wiringPi'. Unable to fetch URL from any source.

ERROR: wiringpi-git-r0 do_fetch: Function failed: base_do_fetch

ERROR: Logfile of failure stored in: /media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/wiringpi/git-r0/temp/log.do_fetch.22117

NOTE: recipe wiringpi-git-r0: task do_fetch: Failed

ERROR: Task (/media/yocto/jenkins/jenkins-meta/jenkins-meta/workspace/a-gateway-os_feature_auto_deploy/lora-gateway-os/layers/bsp/meta-raspberrypi/recipes-devtools/wiringpi/wiringpi_git.bb:do_fetch) failed with exit code '1'
```

Is there a reason this is being installed from Git rather than from the packages as recommended on the installation page at http://wiringpi.com/download-and-install/ ?

Currently, if you try and access git.drogon.com you get the following message:

`git.drogon.net is temporarily unavailable. Please look for alternatives for wiringPi, etc. -Gordon`

proffalken, 2019-08-13:
hmmm, looks like this is a more general Yocto issue, the wiringpi stuff is in https://github.com/agherzan/meta-raspberrypi - logging a bug there too.

proffalken, 2019-08-13:
Further digging, WiringPi is deprecated: http://wiringpi.com/wiringpi-deprecated/

brocaar, 2019-08-14:
I have posted a reply in the meta-raspberrypi issue. I think that is a good location to continue the discussion specifically to wiringpi package.

proffalken, 2019-08-14:
Agree with @brocaar - closing this for now, will re-open if necessary.

closed 2019-08-14 by proffalken

MotschaM, 2019-10-13:
gibt es was neues von wiringpi?