chirpstack/chirpstack-gateway-os #21: AS923 missing for RAK831

jeremyherbert, 2019-09-21:
I have a RAK831 and I have used it on TTN with AS923. However, when I run `gateway-config` AS923 is not one of the options available.

brocaar, 2019-09-23:
If you would like to contribute, adding AS923 configuration would be much appreciated :-)

Please see:
https://github.com/brocaar/lora-gateway-os/tree/master/layers/targets/meta-raspberrypi/recipes-lora/lora-packet-forwarder

And:
https://github.com/brocaar/lora-gateway-os/blob/master/layers/targets/meta-raspberrypi/recipes-core/gateway-config/gateway-config/gateway-config.sh

jeremyherbert, 2019-09-24:
I tried to do that, but #22 blocked me ;)

tonysmith55, 2019-09-24:
Hi Orne,  I created AS923-1 and AS923-2 when the original AU and US frequency blocks were developed.  I'm not confident with git but the files can be found at https://github.com/tonysmith55/lora-gateway-additions  
You will also see files for the other countries currently not included (eg India, Korea, China etc etc)

brocaar, 2020-04-02:
This has been implemented by #41

closed 2020-04-02 by brocaar