chirpstack/chirpstack-gateway-os #22: App server listens by default on IPv6 only

jeremyherbert, 2019-09-21:
Running `netstat -vatn` on a brand new flashed device gives me the following:

```
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 127.0.0.1:6379          0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp        0      0 127.0.0.1:5432          0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN
tcp        0      0 127.0.0.1:2812          0.0.0.0:*               LISTEN
tcp        0      0 :::111                  :::*                    LISTEN
tcp        0      0 :::8080                 :::*                    LISTEN
tcp        0      0 ::1:53                  :::*                    LISTEN
tcp        0      0 :::22                   :::*                    LISTEN
tcp        0      0 ::1:5432                :::*                    LISTEN
tcp        0      0 :::1883                 :::*                    LISTEN
tcp        0      0 :::8000                 :::*                    LISTEN
tcp        0      0 :::8001                 :::*                    LISTEN
tcp        0      0 :::8003                 :::*                    LISTEN
```

Based on this, it is clear that the app server is only running on IPv6. 

It doesn't seem possible to disable IPv6 using sysctl (it just doesn't work) like this:
```
sudo sysctl -w net.ipv6.conf.all.disable_ipv6=1
sudo sysctl -w net.ipv6.conf.default.disable_ipv6=1
```

and if I disable ipv6 using the kernel command line argument `ipv6.disable=1` then SSH doesn't listen properly anymore (actually, nothing does; I can ping the device but it otherwise shows up as having zero open ports using nmap).

Disabling IPv6 kills the HDMI output as well for some reason.

The image I am using is `lora-gateway-os-full-raspberrypi3--20190810092349.sdimg`

jeremyherbert, 2019-09-21:
Looks like something else might be going on. My configuration file explicitly states to listen on `0.0.0.0` but it doesn't seem to change anything (this is `/etc/lora-app-server/lora-app-server.toml`)

```
[general]
log_level=4
password_hash_iterations=1000

[postgresql]
dsn="postgres://loraserver_as:loraserver_as@localhost/loraserver_as?sslmode=disable"

[application_server.external_api]
jwt_secret=""
bind="0.0.0.0:8080"
```
(I've removed the secret)

`wget http://localhost:8080` does indeed work regardless of whether the `bind` line is present in the config file.

brocaar, 2020-01-27:
Is this an actual issue? I believe this has been brought up before, but I believe things do work as expected.

closed 2020-02-24 by brocaar