chirpstack/chirpstack-gateway-os #23: Build does not work on Ubuntu Desktop small version, use Full version

andershedberg, 2019-09-28:
Probably not an issue with lora-gateway-os, but I'm unused to these things and wish to log it somewhere as it might help someone else. (Please close issue and I think it will still be searchable )

Installing a fresh Ubuntu 18.04 LTS or 19.04 Desktop version you are asked for a Full ( with office packages and everything ) or a Small ( with only webbrowser ) installation of Ubuntu. The Small version that ought to be enough fails the build in some openembedded nss-xxx.bb lib-files, that passes fine if built on the Full version (after adding git, docker and docker-composer to both). Very strange as the build system is running inside containers that should contain everything by themselves?!? but this is far beyond my understanding.

Conclusion - Build on Full Desktop version of Ubuntu and the instructions here really works - https://github.com/brocaar/lora-gateway-os/blob/master/README.md ( after applying the fix in #20 if having the WiringPi Error )

solotaker, 2019-10-16:
Hi，
I tried to generate an image file but it failed.
After executing `bitbake lora-gateway-os-base`
Always stop at

> Yocto @ 37458b161cfb:/ build $ bitbake lora-gateway-os-base
> Loading cache: 15%| #############

Then display time out after a very long wait time
My build environment is ubuntu16.04 LTS

> Saber@ubuntu:~$ docker-compose -version
> Docker-compose version 1.24.1, build 4667896b
> Saber@ubuntu:~$ docker version
> Client: Docker Engine - Community
>  Version: 19.03.3
>  API version: 1.40
>  Go version: go1.12.10
>  Git commit: a872fc2
>  Built: Tue Oct 8 00:59:54 2019
>  OS/Arch: linux/amd64
>  Experimental: false
> 
> Server: Docker Engine - Community
>  Engine:
>   Version: 19.03.3
>   API version: 1.40 (minimum version 1.12)
>   Go version: go1.12.10
>   Git commit: a872fc2
>   Built: Tue Oct 8 00:58:28 2019
>   OS/Arch: linux/amd64
>   Experimental: false
>  Containerd:
>   Version: 1.2.10
>   GitCommit: b34a5c8af56e510852c35414db4c1f4fa6172339
>  Runc:
>   Version: 1.0.0-rc8+dev
>   GitCommit: 3e425f80a8c931f88e6d94a8c831b9d5aa481657
>  Docker-init:
>   Version: 0.18.0
>   GitCommit: fec3683
would you give me some advice?

brocaar, 2019-10-21:
@andershedberg thanks, I have not experienced this issue myself. I'm trying to minimize these environment dependent differences by using Docker, but apparently there might still be differences based on setup. These are hard to validate.

@solotaker I don't know, maybe you are running out of memory? I'm not sure if I could help you here. Have you looked at the https://www.yoctoproject.org/ documentation?

closed 2019-10-21 by brocaar