chirpstack/chirpstack-gateway-os #29: The content of chirpstack-hugo-theme cannot be downloaded

mbedded-dev, 2019-12-30:
When I tried the first update of chirpstack-gateway-os submodules:

`make submodules` 

there was a problem during cloning chirpstack-hugo-theme repository:

> git@github.com: Permission denied (publickey).
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.

The problem occurs even if I add RSA key to my known_hosts file. For other submodules/repositories I have not encountered this problem.
@brocaar

piechade, 2020-01-03:
I'm copying the whole config ssh config from my host system in the Docker container.
To add RSA to known_hosts is not enough, you need also a working ssh key that is added to you're github account.

```
COPY docker/.ssh /home/yocto
RUN chmod 700 /home/yocto/.ssh
RUN chmod 644 /home/yocto/.ssh/id_rsa.pub
RUN chmod 600 /home/yocto/.ssh/id_rsa
```

This solution is not really gerate but was working for me.

brocaar, 2020-01-03:
I have updated the submodule URL, this should fix the issue.

closed 2020-01-03 by brocaar

piechade, 2020-01-03:
> I have updated the submodule URL, this should fix the issue.

````
fatal: unable to access 'https://github.com:brocaar/chirpstack-hugo-theme.git/': Port number ended with 'b'
````

commit fd9973f3 (2020-01-03): Update chirpstack-hugo-theme submodule url.

Closes #29.