chirpstack/chirpstack-gateway-os #30: Build image in Docker

piechade, 2020-01-03:
`ERROR: rootfs-overlay-initramfs-1.0-r0 do_rootfs: The postinstall intercept hook 'update_gio_module_cache' failed, details in /build/tmp/raspberrypi/raspberrypi3-glibc/work/raspberrypi3-oe-linux-gnueabi/rootfs-overlay-initramfs/1.0-r0/temp/log.do_rootfs
ERROR: rootfs-overlay-initramfs-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /build/tmp/raspberrypi/raspberrypi3-glibc/work/raspberrypi3-oe-linux-gnueabi/rootfs-overlay-initramfs/1.0-r0/temp/log.do_rootfs.20245
ERROR: Task (/chirpstack-gateway-os/meta/recipes-core/images/rootfs-overlay-initramfs.bb:do_rootfs) failed with exit code '1'`

brocaar, 2020-01-27:
I'm not sure how to reproduce this issue. Are you still experiencing it?

piechade, 2020-02-01:
I switched to my Linux machine and not tried it again on Windows.

brocaar, 2020-02-06:
Ok. I'm closing this issue for now.

closed 2020-02-12 by brocaar