chirpstack/chirpstack-gateway-os #32: Redis database stops working on devices after some time

andershedberg, 2020-01-09:
I escalate this to a bug here at the github repo since I have now found two other people experience this on the user forums, and it is a real deal-breaker for the gateway-os. I.e gateway can not be used as it dies in the field.. 

https://forum.chirpstack.io/t/redis-error-after-gatewayos-reboot/5568 ( other peoples report on LorixOne and RAK2245 )

https://forum.chirpstack.io/t/redis-database-stopped-running-anyone-else-that-has-seen-this/7095 ( my report on Raspberry pi + RisingHF )

As seen in my bootlog, redis does start, but it is dead later on. My linuxskills is not that good so this is as far as I have come with the logs... an error in an append only file?
```
raspberrypi3:~$ sudo more /var/log/boot
Wed Jan  8 13:20:39 2020: Wed Jan  8 13:20:39 UTC 2020
Wed Jan  8 13:20:39 2020: INIT: Entering runlevel: 5
Wed Jan  8 13:20:39 2020: Configuring network interfaces... done.
Wed Jan  8 13:20:39 2020: Starting system message bus: dbus.
Wed Jan  8 13:20:39 2020: Starting Connection Manager
Wed Jan  8 13:20:39 2020: Starting Dropbear SSH server: dropbear.
Wed Jan  8 13:20:40 2020: Starting rpcbind daemon...done.
Wed Jan  8 13:20:40 2020: Starting advanced power management daemon: No APM support in kernel
Wed Jan  8 13:20:40 2020: (failed.)
Wed Jan  8 13:20:40 2020: Starting bluetooth: bluetoothd.
Wed Jan  8 13:20:40 2020: Starting lora-app-server
Wed Jan  8 13:20:40 2020: Starting lora-gateway-bridge
Wed Jan  8 13:20:40 2020: Resetting concentrator board using pin 7
Wed Jan  8 13:20:41 2020: Starting lora-packet-forwarder
Wed Jan  8 13:20:41 2020: Starting loraserver
Wed Jan  8 13:20:41 2020: mender is disabled, update /etc/default/mender
Wed Jan  8 13:20:41 2020: Starting ntpd: done
Wed Jan  8 13:20:42 2020: Starting postgresql-server service:  [ OK ]
Thu Jan  9 22:53:07 2020: Starting syslogd/klogd: done
Thu Jan  9 22:53:07 2020: wireguard is disabled, update /etc/default/wireguard
Thu Jan  9 22:53:07 2020:  * Starting Avahi mDNS/DNS-SD Daemon: avahi-daemon
Thu Jan  9 22:53:07 2020:    ...done.
Thu Jan  9 22:53:07 2020: Starting Telephony daemon
Thu Jan  9 22:53:07 2020: Starting Mosquitto message broker mosquitto
Thu Jan  9 22:53:07 2020: Starting redis-server...
Thu Jan  9 22:53:07 2020: 414:C 09 Jan 22:53:07.709 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
Thu Jan  9 22:53:07 2020: 414:C 09 Jan 22:53:07.709 # Redis version=4.0.8, bits=32, commit=00000000, modified=0, pid=414, just started
Thu Jan  9 22:53:07 2020: 414:C 09 Jan 22:53:07.709 # Configuration loaded
Thu Jan  9 22:53:07 2020: Starting MonitStarting Monit 5.26.0 daemon with http interface at [localhost]:2812
Thu Jan  9 22:53:07 2020: .
raspberrypi3:~$ sudo ps -aux | grep redis
admin     1465  0.0  0.0   2384   476 pts/0    S+   23:06   0:00 grep redis
raspberrypi3:~$
```
```
raspberrypi3:~$ sudo cat /var/log/messages | grep redis | more
Jan  9 22:53:07 raspberrypi3 user.warn redis-server: oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
Jan  9 22:53:07 raspberrypi3 user.warn redis-server: Redis version=4.0.8, bits=32, commit=00000000, modified=0, pid=414, just started
Jan  9 22:53:07 raspberrypi3 user.warn redis-server: Configuration loaded
Jan  9 22:53:07 raspberrypi3 local0.notice redis[415]: Increased maximum number of open files to 10032 (it was originally set to 1024).
Jan  9 22:53:07 raspberrypi3 local0.warn redis[415]: Warning: 32 bit instance detected but no memory limit set. Setting 3 GB maxmemory limit with 'noeviction' policy now.
Jan  9 22:53:07 raspberrypi3 local0.notice redis[415]:                 _._                                                              _.-``__ ''-._                                                    _.-``    `.  `_.  ''-._           Redis 4.0.8 (00000000/0) 32 bit   .-`` .-`
Jan  9 22:53:07 raspberrypi3 local0.warn redis[415]: WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
Jan  9 22:53:07 raspberrypi3 local0.warn redis[415]: Server initialized
Jan  9 22:53:07 raspberrypi3 local0.warn redis[415]: WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memo
Jan  9 22:53:17 raspberrypi3 local0.warn redis[415]: Bad file format reading the append only file: make a backup of your AOF file, then use ./redis-check-aof --fix <filename>
```

andershedberg, 2020-01-10:
Sorry for long gibberish post, **see last lines for how to fix this on the command line.** ( First my notes of where to find files if someone is trying to change the startup scripts )

On my buildserver, the config file for building redis is
```
~/lora-gateway-os/layers/openembedded/meta-openembedded/meta-oe/recipes-extended/redis/redis_4.0.8.bb 
```
and it contains 
```
[...]
USERADD_PARAM_${PN}  = "--system --home-dir /var/lib/redis -g redis --shell /bin/false redis"
[...]
```

```
~/lora-gateway-os/layers/openembedded/meta-openembedded/meta-oe/recipes-extended/redis/redis$ more redis.service
[Unit]
Description=Redis In-Memory Data Store
After=network.target

[Service]
User=redis
Group=redis
ExecStart=/usr/bin/redis-server /etc/redis/redis.conf
ExecStop=/usr/bin/redis-cli shutdown
Restart=always
LimitNOFILE=10032

[Install]
WantedBy=multi-user.target
```
The redis.conf file contains
```
# Redis configuration file example.
#
# Note that in order to read the configuration file, Redis must be
# started with the file path as first argument:
#
# ./redis-server /path/to/redis.conf
[...]
# The working directory.
#
# The DB will be written inside this directory, with the filename specified
# above using the 'dbfilename' configuration directive.
#
# The Append Only File will also be created inside this directory.
#
# Note that you must specify a directory here, not a file name.
dir /var/lib/redis/
[...]
# If the AOF is enabled on startup Redis will load the AOF, that is the file
# with the better durability guarantees.
#
# Please check http://redis.io/topics/persistence for more information.

# OE: changed default to enable this
appendonly yes

# The name of the append only file (default: "appendonly.aof")

appendfilename "appendonly.aof"
[...]
# If aof-load-truncated is set to yes, a truncated AOF file is loaded and
# the Redis server starts emitting a log to inform the user of the event.
# Otherwise if the option is set to no, the server aborts with an error
# and refuses to start. When the option is set to no, the user requires
# to fix the AOF file using the "redis-check-aof" utility before to restart
# the server.
#
# Note that if the AOF file will be found to be corrupted in the middle
# the server will still exit with an error. This option only applies when
# Redis will try to read more data from the AOF file but not enough bytes
# will be found.
aof-load-truncated yes
[...]
``` 

On my raspberry 
```
raspberrypi3:~$ more /etc/init.d/redis-server
#!/bin/sh
#
### BEGIN INIT INFO
# Provides:          redis-server
# Required-Start:    $network
# Required-Stop:     $network
# Default-Start:     S 2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Redis, a key-value store
# Description:       Redis is an open source, advanced key-value store.
#                    http://redis.io
### END INIT INFO

test -f /usr/bin/redis-server || exit 0

ARGS="/etc/redis/redis.conf"

case "$1" in
    start)
        echo "Starting redis-server..."
        start-stop-daemon --start --quiet --exec /usr/bin/redis-server -- $ARGS
        ;;
    stop)
        echo "Stopping redis-server..."
        start-stop-daemon --stop --quiet --exec /usr/bin/redis-server
        ;;
    restart)
        echo "Stopping redis-server..."
        start-stop-daemon --stop --quiet --exec /usr/bin/redis-server
        echo "Starting redis-server..."
        start-stop-daemon --start --quiet --exec /usr/bin/redis-server -- $ARGS
        ;;
    *)
        echo "Usage: /etc/init.d/redis-server {start|stop|restart}"
        exit 1
        ;;
esac

exit 0
``` 
The /etc/redis/redis.conf file is the same as above. 

Append only file is here 
```
raspberrypi3:~$ ls -l /var/lib/redis/appendonly.aof
-rw-r--r--    1 root     root      36100661 Dec 20 12:18 /var/lib/redis/appendonly.aof
```
Running repair tool ends with a prompt...
```
raspberrypi3:~$ sudo /usr/bin/redis-check-aof --fix /var/lib/redis/appendonly.aof
0x         226d889: Expected prefix '*', got: '
AOF analyzed: size=36100661, ok_up_to=36100233, diff=428
This will shrink the AOF from 36100661 bytes, with 428 bytes, to 36100233 bytes
Continue? [y/N]:
```

Running it with <<< 'yes' seems to work automatically. 

**Solve the database error from the command line**
```
raspberrypi3:~$ sudo /usr/bin/redis-check-aof --fix /var/lib/redis/appendonly.aof <<< 'yes'
0x         226d889: Expected prefix '*', got: '
AOF analyzed: size=36100661, ok_up_to=36100233, diff=428
This will shrink the AOF from 36100661 bytes, with 428 bytes, to 36100233 bytes
Continue? [y/N]: Successfully truncated AOF
raspberrypi3:~$
```

After a reboot the system ssems to work ok again and everything starts...
```
raspberrypi3:~$ sudo reboot

The system is going down for reboot NOW! (pts/0) (Fri Jan 10 00:17:13 2020):
raspberrypi3:~$
```

andershedberg, 2020-01-10:
I suspect this changed /etc/init.d/redis-server file will actually work on a failed and rebooted system. It do the right things when rebooting ( redis-check-aof is simply inserted into the start and restart parts of the script ).

```
raspberrypi3:~$ more /etc/init.d/redis-server
#!/bin/sh
#
### BEGIN INIT INFO
# Provides:          redis-server
# Required-Start:    $network
# Required-Stop:     $network
# Default-Start:     S 2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Redis, a key-value store
# Description:       Redis is an open source, advanced key-value store.
#                    http://redis.io
### END INIT INFO

test -f /usr/bin/redis-server || exit 0

ARGS="/etc/redis/redis.conf"

case "$1" in
    start)
        echo "Cleaing up redis append file"
        /usr/bin/redis-check-aof --fix /var/lib/redis/appendonly.aof <<< 'yes'
        echo "Starting redis-server..."
        start-stop-daemon --start --quiet --exec /usr/bin/redis-server -- $ARGS
        ;;
    stop)
        echo "Stopping redis-server..."
        start-stop-daemon --stop --quiet --exec /usr/bin/redis-server
        ;;
    restart)
        echo "Stopping redis-server..."
        start-stop-daemon --stop --quiet --exec /usr/bin/redis-server
        echo "Cleaing up redis append file"
        /usr/bin/redis-check-aof --fix /var/lib/redis/appendonly.aof <<< 'yes'
        echo "Starting redis-server..."
        start-stop-daemon --start --quiet --exec /usr/bin/redis-server -- $ARGS
        ;;
    *)
        echo "Usage: /etc/init.d/redis-server {start|stop|restart}"
        exit 1
        ;;
esac

exit 0
```

brocaar, 2020-01-13:
Thanks! I'm in the process of making some updates and improvements and will look into this before I'll release a new version.

andershedberg, 2020-01-13:
Great! 

I do not know if this is related, but on the Applications->devices->device->Activation tab I now  noted (as devices did not work) that they all lacked decryption keys. ( All were ABP devices, so Device address, Network session key, Application session key was missing on that tab ) Entering the values and then pressing (Re)Avtive fixed this issue. Strangely the devices do exist correctly on the Details and Configuration tabs and they worked correctly on the 5th of december on this same gateway.

brocaar, 2020-01-16:
Yes, no Redis data means no Device Session data.

I'll apply the following patch in the Yocto recipes:

```diff
 ARGS="/etc/redis/redis.conf"

+fix_aof() {
+       if [ -f "/var/lib/redis/appendonly.aof" ] then;
+               echo "Cleaning up Redis append-only file"
+               /usr/bin/redis-check-aof --fix /var/lib/redis/appendonly.aof <<< 'yes'
+       fi
+}
+
 case "$1" in
     start)
        echo "Starting redis-server..."
+       fix_aof
         start-stop-daemon --start --quiet --exec /usr/bin/redis-server -- $ARGS
        ;;
     stop)
@@ -28,6 +36,7 @@
         echo "Stopping redis-server..."
         start-stop-daemon --stop --quiet --exec /usr/bin/redis-server
        echo "Starting redis-server..."
+       fix_aof
         start-stop-daemon --start --quiet --exec /usr/bin/redis-server -- $ARGS
        ;;
     *)
```

brocaar, 2020-01-27:
This has been fixed in the latest master branch.

closed 2020-01-27 by brocaar