chirpstack/chirpstack-gateway-os #35: Can not add network server with 'localhost:8080' but works 127.0.0.1:8080

joau-mms, 2020-02-05:
What happened:
On the Web-UI I wanted to add a NetworkServer. names "testns" and Network-server server "localhost:8080" without any TLS cert and no gateway discovery.
After quite some delay the WebUI itself shows a "context deadline exceeded (code: 2)" within a little overlay. But no Network-server was created
On the other hand, using Network-server server: "127.0.0.1:8080" works fine

What I expected:
that the 'localhost:8080' server works ;)

Additional:
[::1]:8080 also doesn't work

system:
Freshly installed gatewayOS full ( image: raspi3-20200127144146 )

brocaar, 2020-02-06:
You might be mixing up the ports, the NS is listening on port `8000`, not `8080`, that is the web-interface.

joau-mms, 2020-02-06:
Well, I shall not file bug reports at the end of my work day :/

Ok I retested it with a "fresh" brain and  port 8000
1. delete content of "data partition" on separate machine
2. config Concentratord for EU868
3. Delete Organization "ChirpStack" within WebUI and create new Organization named "testorg"
4. Add networkserver with localhost:8000 as suggested on the WebUI:
[image]
Did not work.
5. Change localhost to [::1]:8000 -> won't do the trick
6. Change to 127.0.0.1:8000 -> success

The logfiles did not help. There are just messages, that some idiot tries to add a network server without proper security configuration :)
The messages, that someone tries to setup a NS are shown for all test cases (localhost, ipv6 and ipv4)

PS: I really appreciate your work on chirpstack and also the code quality.

brocaar, 2020-02-07:
I have just tried again with the latest Gateway OS version (3.0.0-test.2), but did not experience any issues when using `localhost:8000` ...

joau-mms, 2020-02-10:
GatewayOS Image published 07.02.2020 (20200206232826) on Raspi3
Again testns with localhost:8000, the first click on "Add Network-Server" did nothing but throwing "context deadline exceeded (code: 2)" at me.
I waited a bit and clicked again and it worked.

Besides I'm only using the vanilla GatewayOS, the only somewhat custom configuration is the network:
Ethernet: Directlink to PC without DNS DHCP oder Internet
Wifi: DHCPv4 +v6 also DNS and internet connection. I access the WebUi via Wifi network

paresy, 2020-02-12:
I had the same issue. 127.0.0.1:8000 did the trick.

brocaar, 2020-02-24:
This will probably be fixed in the next release of the AS and NS. There has been a bugfix in the latest gRPC library version.

gom-dot-sh, 2020-04-06:
There is some progress,
IPv6 ([::1]:8000) works at least! 
"localhost" still won't

self compiled ChipstackOS with sources from 04.04.2020

nathan6343, 2020-04-24:
Nothing but problems when I try.
chirpstack-gateway-os-full-raspberrypi4-20200408114609
context deadline exceeded (code: 2) for localhost:8000 and 127.0.0.1:8000

closed 2021-01-28 by brocaar

DALAA-MedAmine, 2021-03-29:
Try this suoltion : 
You must enter `chirpstack-network-server:8000` as the Network Server `hostname:IP`.

wemaxon, 2021-04-25:
It doesnt work in my network either. I just did a fresh gateway OS install on a Raspberry Pi and I'm always getting the "context deadline exceeded (code 2)" error when trying to add a network-server, regardless what I'd type in the Network-server name and server fields.

Teku16, 2022-05-11:
I am having this issue on (what I think is) the latest pi 3 build `chirpstack-gateway-os-full-raspberrypi3-20210918094417.rootfs.wic` and none of the above-mentioend issues fixed it. Any insight @brocaar ?