chirpstack/chirpstack-gateway-os #36: tcpdump missing from chirpstack-gateway-os-base

mrbubble62, 2020-02-11:
I changed MACHINE to raspberrypi0-wifi, build seems successful, cannot see any errors.
I did have to manually copy image to deploy folder. However I am missing tcpdump from the image. Concentratord and Gateway Bridge are running but I cannot troubleshoot connections.
I don't know where in the build process it;s going wrong.

https://github.com/brocaar/chirpstack-gateway-os/compare/master...mrbubble62:master

mrbubble62, 2020-02-11:
oops found it under /usr/sbin

closed 2020-02-11 by mrbubble62

paresy, 2020-02-12:
I think this should be reopened and fixed. The docs do not state that you need to use the full path: https://www.chirpstack.io/guides/troubleshooting/gateway/

mrbubble62, 2020-02-12:
It works if you su to root.