chirpstack/chirpstack-gateway-os #37: GPS on Pi Zero W

mrbubble62, 2020-02-11:
I updated chirpstack-concentratord config  with
```
[gateway]
gps_tty_path="/dev/ttyS0"
```
However still tries to use ttyAMA0
```
Feb 11 23:12:00 gw1 user.info chirpstack-concentratord-sx1301[522]: Starting GPS validation loop
Feb 11 23:12:00 gw1 user.debug chirpstack-concentratord-sx1301[522]: Starting stats loop, stats_interval: 30s
Feb 11 23:12:00 gw1 user.debug chirpstack-concentratord-sx1301[522]: Starting GPS loop
Feb 11 23:12:00 gw1 user.info chirpstack-concentratord-sx1301[522]: GPS TTY port opened for GPS synchronization, gps_tty_path: /dev/ttyAMA0
Feb 11 23:12:00 gw1 user.debug chirpstack-concentratord-sx1301[522]: Current concentrator count_us: 2576389
Feb 11 23:12:00 gw1 user.debug chirpstack-concentratord-sx1301[522]: Concentrator drift, drift_us: -12
Feb 11 23:12:00 gw1 user.debug chirpstack-concentratord-sx1301[522]: Enabling GPS mode for concentrator counter
Feb 11 23:12:00 gw1 user.debug chirpstack-concentratord-sx1301[522]: Starting beacon loop
Feb 11 23:12:00 gw1 user.debug chirpstack-concentratord-sx1301[522]: No beacon frequencies configured, disabling beaconing
Feb 11 23:12:01 gw1 user.warn chirpstack-concentratord-sx1301[522]: GPS time reference is not valid, age: 1581462721.831991473s
Feb 11 23:12:02 gw1 user.warn chirpstack-concentratord-sx1301[522]: GPS time reference is not valid, age: 1581462722.832791572s
```

mrbubble62, 2020-02-13:
I built the firmware image to force ttyS0. (I made the change in chirpstack-concentratord/chirpstack-concentratord-sx1301/src/config/vendor/generic/us915.rs) 
But the concentratord is not happy
```
Enabling GPS mode for concentrator counter
get gps coordinates failed, error: lgw_gps_get failed
GPS time reference is not valid, age: 1581597658.106345851s
GPS coordinates sync completed, coords: Coordinates { latitude: 38.92504168333333, longitude: -77.06265346666666, altitude: -96 }, coords_error: Coordinates { latitude: 0.0, longitude: 0.0, altitud
```
I think the GPS data is getting corrupted due to the software UART.  I will have to go back to getting ttyAMA0 working if I want GPS support.

brocaar, 2020-02-24:
Closing this issue as it also has been reported here: https://github.com/brocaar/chirpstack-gateway-os/issues/37.

closed 2020-02-24 by brocaar