chirpstack/chirpstack-gateway-os #38: gateway-os-base image missing /opt directory

mrbubble62, 2020-02-23:
It looks like there should be gateway-utils installed to /opt/libloragw-sx1301/gateway-utils/
However /opt directory is not in the gateway-os-base image

brocaar, 2020-02-24:
Only if the `-utils` package is part of the ChirpStack Gateway OS images, which is not the case. You can include this package in the `chirpstack-gateway-os-base` and / or `-full` image if you want to use these tools, but you need build your own image.

(https://github.com/brocaar/chirpstack-gateway-os/blob/master/layers/chirpstack/meta-semtech/recipes-lora/libloragw-sx1301/libloragw-sx1301_5.0.1.bb#L50)

closed 2020-02-24 by brocaar

mrbubble62, 2020-02-24:
Sorry I am new to bitbake, should I add this to chirpstack-gateway-os/meta/recipes-core/images/chirpstack-gateway-os-base.bb?

>  IMAGE_INSTALL += " \\
>    gateway-utils \\
>   packagegroup-base \\
>   ...