chirpstack/chirpstack-gateway-os #40: Applications / testapp / HTTP results in empty POST Messages

joau-mms, 2020-02-24:
HTTP Post seems to be empty.
Config:
[image]

Target of HTTP Post is an Apache2 with PHP
```if (isset($_POST)) {
    $text = var_export($_POST, true) . "\n\n";
    file_put_contents("dumb.txt", " " . $text, FILE_APPEND);
    echo print_r($_POST, true);
}
```

Result is dumb.txt containing an empty array:
```
 array (
)
```

brocaar, 2020-02-24:
I don't know your application / setup, but please try to reproduce this with for example https://postb.in/.

joau-mms, 2020-02-25:
Well Thanks!
It was my implementation, is seems, that PHPs $_Post won't fetch json.

closed 2020-02-25 by joau-mms