chirpstack/chirpstack-gateway-os #42: IMST Lite Gateway shield config not working

piechade, 2020-03-14:
I’m trying to get Gateway OS running on an IMST Lite Gateway with Raspberry PI 1.
The “chirpstack-concentratord” is crashing continuously.

`Mar 12 20:33:31 raspberrypi user.err chirpstack-gateway-bridge[751]: time="2020-03-12T20:33:31Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial unix /tmp/co`

brocaar, 2020-03-16:
Have you configured the concentrator properly using the `gateway-config` utility? Could you provide any log output of `chirpstack-concentratord`?

piechade, 2020-03-17:
I used the gateway-config toll and the output of chirpstack-concentratord is:

`raspberrypi:~$ sudo tail -f /var/log/messages |grep chirpstack-concentratord
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 868500000, rf_chain: 1, if_freq: 0
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 867100000, rf_chain: 0, if_freq: -400000
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Configuring multi-SF LoRa channel, channel: 4, enabled: true, freq: 867300000, rf_chain: 0, if_freq: -200000
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Configuring multi-SF LoRa channel, channel: 5, enabled: true, freq: 867500000, rf_chain: 0, if_freq: 0
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Configuring multi-SF LoRa channel, channel: 6, enabled: true, freq: 867700000, rf_chain: 0, if_freq: 200000
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 867900000, rf_chain: 0, if_freq: 400000
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Configuring Std LoRa channel, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Configuring FSK channel, enabled: true, freq: 868800000, rf_chain: 1, if_freq: 300000
Mar 17 09:13:15 raspberrypi user.info chirpstack-concentratord-sx1301[846]: Starting the the concentrator
Mar 17 09:13:25 raspberrypi user.err monit[463]: 'chirpstack-concentratord' process is not running
Mar 17 09:13:25 raspberrypi user.info monit[463]: 'chirpstack-concentratord' trying to restart
Mar 17 09:13:25 raspberrypi user.info monit[463]: 'chirpstack-concentratord' start: '/etc/init.d/chirpstack-concentratord start'
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Starting Concentratord SX1301 (version: 3.0.0-test.5, docs: https://www.chirpstack.io/concentratord/)
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Setting spi device path, spidev_path: /dev/spidev0.0
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Setting board configuration, lorawan_public: true, clock_source: 1
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Setting up concentrator radios
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring radio, radio: 0, enabled: true, center_freq: 867500000, type: SX1257
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring radio, radio: 1, enabled: true, center_freq: 868500000, type: SX1257
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Setting up concentrator channels
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring multi-SF LoRa channel, channel: 0, enabled: true, freq: 868100000, rf_chain: 1, if_freq: -400000
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring multi-SF LoRa channel, channel: 1, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 868500000, rf_chain: 1, if_freq: 0
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 867100000, rf_chain: 0, if_freq: -400000
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring multi-SF LoRa channel, channel: 4, enabled: true, freq: 867300000, rf_chain: 0, if_freq: -200000
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring multi-SF LoRa channel, channel: 5, enabled: true, freq: 867500000, rf_chain: 0, if_freq: 0
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring multi-SF LoRa channel, channel: 6, enabled: true, freq: 867700000, rf_chain: 0, if_freq: 200000
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 867900000, rf_chain: 0, if_freq: 400000
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring Std LoRa channel, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Configuring FSK channel, enabled: true, freq: 868800000, rf_chain: 1, if_freq: 300000
Mar 17 09:13:26 raspberrypi user.info chirpstack-concentratord-sx1301[858]: Starting the the concentrator
Mar 17 09:13:37 raspberrypi user.err monit[463]: 'chirpstack-concentratord' process is not running
Mar 17 09:13:37 raspberrypi user.info monit[463]: 'chirpstack-concentratord' trying to restart
Mar 17 09:13:37 raspberrypi user.info monit[463]: 'chirpstack-concentratord' start: '/etc/init.d/chirpstack-concentratord start'
Mar 17 09:13:37 raspberrypi user.info chirpstack-concentratord-sx1301[871]: Starting Concentratord SX1301 (version: 3.0.0-test.5, docs: https://www.chirpstack.io/concentratord/)
[... cut ...]
Mar 17 09:13:37 raspberrypi user.info chirpstack-concentratord-sx1301[871]: Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 867900000, rf_chain: 0, if_freq: 400000
Mar 17 09:13:37 raspberrypi user.info chirpstack-concentratord-sx1301[871]: Configuring Std LoRa channel, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Mar 17 09:13:37 raspberrypi user.info chirpstack-concentratord-sx1301[871]: Configuring FSK channel, enabled: true, freq: 868800000, rf_chain: 1, if_freq: 300000
Mar 17 09:13:37 raspberrypi user.info chirpstack-concentratord-sx1301[871]: Starting the the concentrator`

brocaar, 2020-03-23:
I have posted some debugging instructions here: https://forum.chirpstack.io/t/imst-lite-gateway-setup/7744/6. Could you test with these commands?

closed 2020-09-05 by piechade