chirpstack/chirpstack-gateway-os #46: High wifi latency and very high jitter

gom-dot-sh, 2020-03-30:
ChirpstackOS-full is used, update 27.03.2020 was installed

Wifi latency and jitter is too high, sometimes exceeding 1000ms, which could be problematic if one needs to hit timed receiving windows of Class A/B devices.

screenshot: 
on the left is the ChipstackOS and on the right side raspbian. Both devices are Model 3 Raspberries and are using the same wifi, placed within 30cm of the wifi access point.
[image]
```
--- 192.168.128.71 ping statistics ---
1325 packets transmitted, 1119 received, +139 errors, 15,5472% packet loss, time 1329944ms  
rtt min/avg/max/mdev = 1.473/317.005/1236.734/300.494 ms

-- 192.168.128.72 ping statistics ---
4945 packets transmitted, 4944 received, 0,0202224% packet loss, time 4949597ms
rtt min/avg/max/mdev = 0.804/4.940/166.763/6.950 ms
```
You can ignore the package losses, that was my doing (reboot of ChirpOS).

brocaar, 2020-03-31:
I can not reproduce your results:

```
   ________    _           _____ __             __     _     
  / ____/ /_  (_)________ / ___// /_____ ______/ /__  (_)___ 
 / /   / __ \/ / ___/ __ \\__ \/ __/ __ `/ ___/ //_/ / / __ \
/ /___/ / / / / /  / /_/ /__/ / /_/ /_/ / /__/ ,< _ / / /_/ /
\____/_/ /_/_/_/  / .___/____/\__/\__,_/\___/_/|_(_)_/\____/ 
                 /_/

Documentation and copyright information:
> www.chirpstack.io

Commands:
> sudo gateway-config  - configure the gateway
> sudo monit summary   - display service monitor

raspberrypi3:~$ ping 192.168.1.150
PING 192.168.1.150 (192.168.1.150): 56 data bytes
64 bytes from 192.168.1.150: seq=0 ttl=64 time=3.723 ms
64 bytes from 192.168.1.150: seq=1 ttl=64 time=6.834 ms
64 bytes from 192.168.1.150: seq=2 ttl=64 time=7.922 ms
64 bytes from 192.168.1.150: seq=3 ttl=64 time=5.815 ms
64 bytes from 192.168.1.150: seq=4 ttl=64 time=6.352 ms
64 bytes from 192.168.1.150: seq=5 ttl=64 time=5.907 ms
64 bytes from 192.168.1.150: seq=6 ttl=64 time=6.241 ms
64 bytes from 192.168.1.150: seq=7 ttl=64 time=6.364 ms
64 bytes from 192.168.1.150: seq=8 ttl=64 time=6.162 ms
64 bytes from 192.168.1.150: seq=9 ttl=64 time=6.450 ms
64 bytes from 192.168.1.150: seq=10 ttl=64 time=5.989 ms
64 bytes from 192.168.1.150: seq=11 ttl=64 time=6.398 ms
64 bytes from 192.168.1.150: seq=12 ttl=64 time=6.113 ms
64 bytes from 192.168.1.150: seq=13 ttl=64 time=6.163 ms
64 bytes from 192.168.1.150: seq=14 ttl=64 time=5.910 ms
64 bytes from 192.168.1.150: seq=15 ttl=64 time=9.330 ms
64 bytes from 192.168.1.150: seq=16 ttl=64 time=7.049 ms
64 bytes from 192.168.1.150: seq=17 ttl=64 time=6.399 ms
64 bytes from 192.168.1.150: seq=18 ttl=64 time=6.093 ms
64 bytes from 192.168.1.150: seq=19 ttl=64 time=5.951 ms
64 bytes from 192.168.1.150: seq=20 ttl=64 time=5.944 ms
64 bytes from 192.168.1.150: seq=21 ttl=64 time=5.988 ms
64 bytes from 192.168.1.150: seq=22 ttl=64 time=5.974 ms
64 bytes from 192.168.1.150: seq=23 ttl=64 time=6.021 ms
64 bytes from 192.168.1.150: seq=24 ttl=64 time=7.562 ms
64 bytes from 192.168.1.150: seq=25 ttl=64 time=6.211 ms
64 bytes from 192.168.1.150: seq=26 ttl=64 time=6.417 ms
64 bytes from 192.168.1.150: seq=27 ttl=64 time=5.798 ms
64 bytes from 192.168.1.150: seq=28 ttl=64 time=28.015 ms
64 bytes from 192.168.1.150: seq=29 ttl=64 time=6.439 ms
64 bytes from 192.168.1.150: seq=30 ttl=64 time=10.823 ms
64 bytes from 192.168.1.150: seq=31 ttl=64 time=6.924 ms
64 bytes from 192.168.1.150: seq=32 ttl=64 time=6.016 ms
64 bytes from 192.168.1.150: seq=33 ttl=64 time=7.565 ms
64 bytes from 192.168.1.150: seq=34 ttl=64 time=7.579 ms
64 bytes from 192.168.1.150: seq=35 ttl=64 time=6.131 ms
64 bytes from 192.168.1.150: seq=36 ttl=64 time=5.899 ms
64 bytes from 192.168.1.150: seq=37 ttl=64 time=5.995 ms
64 bytes from 192.168.1.150: seq=38 ttl=64 time=6.005 ms
64 bytes from 192.168.1.150: seq=39 ttl=64 time=5.807 ms
64 bytes from 192.168.1.150: seq=40 ttl=64 time=7.122 ms
64 bytes from 192.168.1.150: seq=41 ttl=64 time=7.654 ms
64 bytes from 192.168.1.150: seq=42 ttl=64 time=5.841 ms
64 bytes from 192.168.1.150: seq=43 ttl=64 time=5.944 ms
64 bytes from 192.168.1.150: seq=44 ttl=64 time=5.730 ms
^C
--- 192.168.1.150 ping statistics ---
45 packets transmitted, 45 packets received, 0% packet loss
round-trip min/avg/max = 3.723/6.945/28.015 ms
```

```
raspberrypi3:~$ sudo su
Password: 
raspberrypi3:/home/admin# connmanctl 
connmanctl> services
*AO XXX wifi_XXX_managed_psk
connmanctl> services wifi_XXX_managed_psk
/net/connman/service/wifi_XXX_managed_psk
  Type = wifi
  Security = [ psk, wps ]
  State = online
  Strength = 68
  Favorite = True
  Immutable = True
  AutoConnect = True
  Name = XXX
  Ethernet = [ Method=auto, Interface=wlan0, Address=XXX, MTU=1500 ]
  IPv4 = [ Method=dhcp, Address=192.168.1.152, Netmask=255.255.255.0, Gateway=192.168.1.1 ]
  IPv4.Configuration = [ Method=dhcp ]
  IPv6 = [ Method=auto, Address=XXX, PrefixLength=64, Privacy=disabled ]
  IPv6.Configuration = [ Method=auto, Privacy=disabled ]
  Nameservers = [ 192.168.1.1, XXX ]
  Nameservers.Configuration = [  ]
  Timeservers = [  ]
  Timeservers.Configuration = [  ]
  Domains = [ home ]
  Domains.Configuration = [  ]
  Proxy = [ Method=direct ]
  Proxy.Configuration = [  ]
  mDNS = False
  mDNS.Configuration = False
  Provider = [  ]
```

gom-dot-sh, 2020-03-31:
Well, that's odd.

```
connmanctl> services wifi_xxx_managed_psk
/net/connman/service/wifi_xxx_managed_psk
  Type = wifi
  Security = [ psk ]
  State = online
  Strength = 83
  Favorite = True
  Immutable = True
  AutoConnect = True
  Name = raspberryjam
  Ethernet = [ Method=auto, Interface=wlan0, Address= mac, MTU=1500 ]
  IPv4 = [ Method=dhcp, Address=192.168.128.71, Netmask=255.255.255.0, Gateway=192.168.128.32 ]
  IPv4.Configuration = [ Method=dhcp ]
  IPv6 = [ Method=auto, Address=2003:..:1c35, PrefixLength=64, Privacy=disabled ]
  IPv6.Configuration = [ Method=auto, Privacy=disabled ]
  Nameservers = [ 192.168.128.1, 145.253.2.11, fe80::1 ]
  Nameservers.Configuration = [  ]
  Timeservers = [  ]
  Timeservers.Configuration = [  ]
  Domains = [ dom.lokal ]
  Domains.Configuration = [  ]
  Proxy = [ Method=direct ]
  Proxy.Configuration = [  ]
  mDNS = False
  mDNS.Configuration = False
  Provider = [  ]
```

To get a latency below 10ms I must use Ethernet. Albeit ~5ms latency for cabled network with only one switch in between is also quite high. All other Raspberries on the same switch are having >1ms
```
--- 192.168.128.73 ping statistics ---
393 packets transmitted, 393 received, 0% packet loss, time 392494ms
rtt min/avg/max/mdev = 1.029/5.136/11.533/0.699 ms
```

Detailed Configuration:
Raspi3 with RAK2245 Pi HAT
ChripStack is configured with one NetworkServer, Gateway, Organzation, Application and one Device (which is offline while testing the network).

I can change the Raspi and Concentrator (Dragino PG1301) but It's quite likely I need to hack some of your Code to get the PG1301 working and recompile the whole GatewayOS. Please mention if this could help you.

gom-dot-sh, 2020-04-06:
Hello again,
I tested again and used a self compiled chirpstackOS-full (git pull around 04.April.2020 20:00 UTC), switched to the other RaspberryPi3 and also a new wifi (Android Hotspot). The results are somewhat consistent. Latencies with ChirpstackOS are bad and jitter is worse (4/150/500 min/mean/max) while raspbian performance is normal.

brocaar, 2023-12-22:
I'm closing this issue as I'm not sure if it is still relevant after Yocto -> OpenWrt migration.

closed 2023-12-22 by brocaar