chirpstack/chirpstack-gateway-os #59: Chirpstack OS building error

George-Ng2020, 2020-08-09:
Hi 

I already installed docker compose to ubuntu 18.04, however I follow the step of setting up I got the below error:

george@george-ThinkPad-X200:~$ make submodules

Command 'make' not found, but can be installed with:

sudo apt install make      
sudo apt install make-guile

george@george-ThinkPad-X200:~$ make submodules

Command 'make' not found, but can be installed with:

sudo apt install make      
sudo apt install make-guile

george@george-ThinkPad-X200:~$ docker-compose run --rm busybox
ERROR: 
        Can't find a suitable configuration file in this directory or any
        parent. Are you in the right directory?

        Supported filenames: docker-compose.yml, docker-compose.yaml
        
george@george-ThinkPad-X200:~$ 

What other steps I need to do before I can run docker-compose run --rm busybox?

andershedberg, 2020-08-24:
When I built it last year, there were some dependencies that required the docker containers to run on a certain os-version if I remember correctly so I had to set up a separate virtual machine for it. I though docker would remove such things, but apparently not. Now I'm going to try to build the latest version on a Windows10 machine to see how it goes.

andershedberg, 2020-09-01:
@George-Ng2020 Did you see this - https://github.com/brocaar/chirpstack-gateway-os/issues/23 ( I.e you need the full version of ubuntu 18.04, the smaller version does not cut it )

andershedberg, 2020-09-02:
> george@george-ThinkPad-X200:~$ make submodules
> 
> Command 'make' not found, but can be installed with:
> 
> sudo apt install make

The Ubuntu 20.04LTS version does not have the tool "make" installed. You need to install it on your computer, just type in
  
```sudo apt install make``` 

in the terminal and hit enter and let it install. Then repeat the steps that could not be executed.

brocaar, 2020-09-08:
I think the errors are self explaining?

```
george@george-ThinkPad-X200:~$ make submodules

Command 'make' not found, but can be installed with:

sudo apt install make
sudo apt install make-guile
```

You have to install `make` using the mentioned command :)

```
george@george-ThinkPad-X200:~$ docker-compose run --rm busybox
ERROR:
Can't find a suitable configuration file in this directory or any
parent. Are you in the right directory?
```

You have to execute the command within the repository that you have cloned (which contains the `docker-compose.yml` file).

closed 2020-09-08 by brocaar