chirpstack/chirpstack-gateway-os #65: Framebuffer DT overlay is set twice

piechade, 2020-10-07:
There is a issue with the device tree on the latest kernel for pi 1, somewhere a framebuffer is declared twice.
My issue that i created about it: https://github.com/agherzan/meta-raspberrypi/issues/710

## Steps to reproduce this issue

Steps:

1. Build the image for pi 1
2. boot

## Log output

`/var/log/dmesg`

```shell
[    1.511411] simple-framebuffer 1e513000.framebuffer: framebuffer at 0x1e513000, 0x6d8c00 bytes, mapped to 0x(ptrval)
[    1.511493] simple-framebuffer 1e513000.framebuffer: format=a8r8g8b8, mode=1824x984x32, linelength=7296
[    1.547602] Console: switching to colour frame buffer device 228x61
[    1.580097] simple-framebuffer 1e513000.framebuffer: fb0: simplefb registered!
[    1.581034] sysfs: cannot create duplicate filename '/devices/platform/1e513000.framebuffer'
[    1.581342] CPU: 0 PID: 1 Comm: swapper Not tainted 5.4.59 #1
[    1.581563] Hardware name: BCM2835
[    1.581706] Backtrace:
[    1.581853] [<c00157dc>] (dump_backtrace) from [<c0015b28>] (show_stack+0x20/0x24)
[    1.582139]  r7:00000000 r6:cb5d87a0 r5:cb51e5a0 r4:cb47b000
[    1.582376] [<c0015b08>] (show_stack) from [<c07bc1b4>] (dump_stack+0x20/0x28)
[    1.582664] [<c07bc194>] (dump_stack) from [<c0252018>] (sysfs_warn_dup+0x60/0x74)
[    1.582960] [<c0251fb8>] (sysfs_warn_dup) from [<c025216c>] (sysfs_create_dir_ns+0xec/0x10c)
[    1.583267]  r7:00000000 r6:cb51e5a0 r5:ffffffef r4:cb5e2a10
[    1.583510] [<c0252080>] (sysfs_create_dir_ns) from [<c07c14e8>] (kobject_add_internal+0x90/0x308)
[    1.583832]  r6:c0fed928 r5:c0fed928 r4:cb5e2a10
[    1.584032] [<c07c1458>] (kobject_add_internal) from [<c07c1c98>] (kobject_add+0x60/0xc4)
[    1.584333]  r9:c0ac282c r8:c102c000 r7:00000000 r6:c0fed928 r5:cb5e2a10 r4:00000000
[    1.584649] [<c07c1c38>] (kobject_add) from [<c0523a78>] (device_add+0x100/0x61c)
[    1.584923]  r3:00000000 r2:00000000
[    1.585077]  r6:c0fed928 r5:dbf62b9c r4:cb5e2a10
[    1.585280] [<c0523978>] (device_add) from [<c0648c10>] (of_device_add+0x40/0x48)
[    1.585663]  r8:c102c000 r7:00000000 r6:00000000 r5:dbf62b9c r4:cb5e2a00
[    1.585943] [<c0648bd0>] (of_device_add) from [<c0649398>] (of_platform_device_create_pdata+0xb4/0xfc)
[    1.586289] [<c06492e4>] (of_platform_device_create_pdata) from [<c0649400>] (of_platform_device_create+0x20/0x24)
[    1.586653]  r7:c09c31e0 r6:c10d28f0 r5:00000000 r4:dbf62b9c
[    1.586896] [<c06493e0>] (of_platform_device_create) from [<c0aa366c>] (simplefb_init+0x74/0x9c)
[    1.587228] [<c0aa35f8>] (simplefb_init) from [<c000ade0>] (do_one_initcall+0x50/0x1d4)
[    1.587523]  r7:00000000 r6:c102c000 r5:c0ac284c r4:c0aa35f8
[    1.598416] [<c000ad90>] (do_one_initcall) from [<c0a731c0>] (kernel_init_freeable+0x130/0x1f4)
[    1.609505]  r7:c0a6f4a0 r6:c102c000 r5:c0ac284c r4:c0aeb94c
[    1.620384] [<c0a73090>] (kernel_init_freeable) from [<c07d3af0>] (kernel_init+0x18/0xfc)
[    1.631302]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c07d3ad8
[    1.642452]  r4:00000000
[    1.653346] [<c07d3ad8>] (kernel_init) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[    1.664365] Exception stack(0xcb4cbfb0 to 0xcb4cbff8)
[    1.675455] bfa0:                                     00000000 00000000 00000000 00000000
[    1.686600] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.697712] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    1.708843]  r5:c07d3ad8 r4:00000000
[    1.719749] kobject_add_internal failed for 1e513000.framebuffer with -EEXIST, don't try to register things with the same name in the same directory.
```

## Your Environment

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | v3.3.2  |

brocaar, 2020-10-07:
@piechade does this error also occurs with the last master branch version? Because https://github.com/brocaar/chirpstack-gateway-os/commit/a958ac06742fbeec67688c684dcd54a1df4f5f41 updates the meta-raspberrypi layer:

> Submodule meta-raspberrypi updated from ac7174 to 085fb0

piechade, 2020-10-07:
> @piechade does this error also occurs with the last master branch version? Because [a958ac0](https://github.com/brocaar/chirpstack-gateway-os/commit/a958ac06742fbeec67688c684dcd54a1df4f5f41) updates the meta-raspberrypi layer:
> 
> > Submodule meta-raspberrypi updated from ac7174 to 085fb0

Exclusively with the latest commit.

piechade, 2020-10-08:
@brocaar something new? Can I help with something?

brocaar, 2020-10-14:
@piechade I haven't had the time to look into this yet. As far as I know, there is no config in the ChirpStack Gateway OS recipes that defines this duplicate frame-buffer config for the Pi 1, so I think we must look into the `meta-raspberrypi` config or `u-boot` config related to the Pi 1 machine.

I will try to reproduce this asap myself on a Pi 1, but I think my guess would be as good as yours at this stage. So any help is appreciated :)

piechade, 2020-10-20:
> @piechade I haven't had the time to look into this yet. As far as I know, there is no config in the ChirpStack Gateway OS recipes that defines this duplicate frame-buffer config for the Pi 1, so I [...]

Did you have time to check it?

brocaar, 2020-10-21:
Not yet, but I'm also waiting for https://github.com/openembedded/meta-openembedded/issues/285 to be resolved. Because of that issue, I can no longer build the wireguard module. According to the latest update, that issue should be fixed by this week. After that, I'll update all dependencies again.

piechade, 2021-01-02:
@brocaar News about the issue? Is it fixed in the latest version?

brocaar, 2021-01-06:
It was not fixed, but in the end I created a work-around  and released v3.3.3. Could you test again with this version?

piechade, 2021-01-07:
The other issue was my fault, sorry about that.
The framebuffer error is the same as before, only the error addresses changed.

brocaar, 2023-12-22:
I'm closing this issue as I'm not sure if it is still relevant after Yocto -> OpenWrt migration.

closed 2023-12-22 by brocaar