chirpstack/chirpstack-gateway-os #66: Sandbox Lorago Port doesn't work with latest release of chirpstack

thugbart, 2020-11-07:
- [X] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?
The lorago shield doesn't work with latest version. No leds are blinking

## What did you expect?

## Steps to reproduce this issue

Steps:

1. Install and plug the sandbox lorago port shielf
2.
3.
4.

## Could you share your log output?

```shell

```

## Your Environment
Raspberry pi 4

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | v?.?.?  |

landersson936, 2021-06-30:
any resolution to this?
I have the same problem.

brocaar, 2021-09-15:
I have tested this shield against the latest master version (https://github.com/brocaar/chirpstack-gateway-os/commit/82e46e6b1897cdd71f577c61990630ce7633f0cf), and I can (no longer) reproduce the issue, the shield works fine.

closed 2021-09-15 by brocaar

thugbart, 2021-11-17:
Hi,

  I've tested again (With 3.5.1) with the sandbox lorago port both on raspberry pi 3 and 4 and still it doesn't work. What pin number did you used for reset ?

Regards

whotwagner, 2022-01-28:
I also have issues with the latest LoraGo Port and with chirpstack-gateway-os-full 3.5.1. For some reasons I get the following log-messages:

```
Jan 28 16:40:56 raspberrypi4 user.err monit[490]: 'chirpstack-concentratord' process is not running
Jan 28 16:40:56 raspberrypi4 user.info monit[490]: 'chirpstack-concentratord' trying to restart
Jan 28 16:40:56 raspberrypi4 user.info monit[490]: 'chirpstack-concentratord' start: '/etc/init.d/chirpstack-concentratord start'
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring reset pin, chip: 0, pin: 0
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Starting Concentratord SX1301 (version: 3.3.1, docs: https://www.chirpstack.io/concentratord/)
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Triggering concentrator reset
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Setting spi device path, spidev_path: /dev/spidev0.0
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Setting board configuration, lorawan_public: true, clock_source: 1
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Setting up concentrator radios
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring radio, radio: 0, enabled: true, center_freq: 867500000, type: SX1257
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring radio, radio: 1, enabled: true, center_freq: 868500000, type: SX1257
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Setting up concentrator channels
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring multi-SF LoRa channel, channel: 0, enabled: true, freq: 868100000, rf_chain: 1, if_freq: -400000
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring multi-SF LoRa channel, channel: 1, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 868500000, rf_chain: 1, if_freq: 0
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 867100000, rf_chain: 0, if_freq: -400000
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring multi-SF LoRa channel, channel: 4, enabled: true, freq: 867300000, rf_chain: 0, if_freq: -200000
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring multi-SF LoRa channel, channel: 5, enabled: true, freq: 867500000, rf_chain: 0, if_freq: 0
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring multi-SF LoRa channel, channel: 6, enabled: true, freq: 867700000, rf_chain: 0, if_freq: 200000
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 867900000, rf_chain: 0, if_freq: 400000
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring Std LoRa channel, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Configuring FSK channel, enabled: true, freq: 868800000, rf_chain: 1, if_freq: 300000
Jan 28 16:40:56 raspberrypi4 user.info chirpstack-concentratord-sx1301[741]: Starting the concentrator
Jan 28 16:40:58 raspberrypi4 user.err chirpstack-gateway-bridge[503]: time="2022-01-28T16:40:58.237105704Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial un
Jan 28 16:41:01 raspberrypi4 user.err chirpstack-gateway-bridge[503]: time="2022-01-28T16:41:01.747990972Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial un
Jan 28 16:41:05 raspberrypi4 user.err chirpstack-gateway-bridge[503]: time="2022-01-28T16:41:05.258217555Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial un
Jan 28 16:41:06 raspberrypi4 user.err monit[490]: 'chirpstack-concentratord' process is not running
Jan 28 16:41:06 raspberrypi4 user.info monit[490]: 'chirpstack-concentratord' trying to restart
Jan 28 16:41:06 raspberrypi4 user.info monit[490]: 'chirpstack-concentratord' start: '/etc/init.d/chirpstack-concentratord start'
Jan 28 16:41:06 raspberrypi4 user.info chirpstack-concentratord-sx1301[746]: Configuring reset pin, chip: 0, pin: 0
Jan 28 16:41:06 raspberrypi4 user.info chirpstack-concentratord-sx1301[746]: Starting Concentratord SX1301 (version: 3.3.1, docs: https://www.chirpstack.io/concentratord/)
Jan 28 16:41:06 raspberrypi4 user.info chirpstack-concentratord-sx1301[746]: Triggering concentrator reset
Jan 28 16:41:07 raspberrypi4 user.info chirpstack-concentratord-sx1301[746]: Setting spi device path, spidev_path: /dev/spidev0.0
Jan 28 16:41:07 raspberrypi4 user.info chirpstack-concentratord-sx1301[746]: Setting board configuration, lorawan_public: true, clock_source: 1
[... cut ...]
Jan 28 16:41:07 raspberrypi4 user.info chirpstack-concentratord-sx1301[746]: Configuring multi-SF LoRa channel, channel: 1, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Jan 28 16:41:07 raspberrypi4 user.info chirpstack-concentratord-sx1301[746]: Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 868500000, rf_chain: 1, if_freq: 0
Jan 28 16:41:07 raspberrypi4 user.info chirpstack-concentratord-sx1301[746]: Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 867100000, rf_chain: 0, if_freq: -400000
...
...
```

mjohanning99, 2022-03-21:
I can confirm that it is not working on my install either.