chirpstack/chirpstack-gateway-os #67: Chirpstack concentratord building error

SLDP-8255, 2020-12-11:
Hello,
I've encountered the problem with concentratord package during chirpstack-gateway-os-full build process. I was building the image on Ubuntu 18.04 LTS, all necessary packages and toolchain for Yocto Project already installed. It looks like some dependency error for rustc. 

[Chirpstack Gateway OS revision](https://github.com/brocaar/chirpstack-gateway-os/commit/be08f86434e25eb94804e02945d6c468ff6bfacc)

The logs fragment regarding the issue:
```
|   running: "/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/wrappers/cc-native-wrapper.sh" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-O2" "-pipe" "-g" "-feliminate-unused-debug-types" "-fmacro-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot=" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot-native=" "-o" "/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/target/release/build/libloading-8b4570a569a949e1/out/src/os/unix/global_static.o" "-c" "src/os/unix/global_static.c"
|   cargo:warning=gcc: error: unrecognized command line option ‘-fmacro-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1’
|   exit code: 1
| 
|   --- stderr
| 
| 
|   error occurred: Command "/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/wrappers/cc-native-wrapper.sh" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-O2" "-pipe" "-g" "-feliminate-unused-debug-types" "-fmacro-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot=" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot-native=" "-o" "/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/target/release/build/libloading-8b4570a569a949e1/out/src/os/unix/global_static.o" "-c" "src/os/unix/global_static.c" with args "cc-native-wrapper.sh" did not execute successfully (status code exit code: 1).
| 
| 
| warning: build failed, waiting for other jobs to finish...
| error: build failed
| WARNING: exit code 101 from a shell command.
| ERROR: Execution of '/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/temp/run.do_compile.22309' failed with exit code 101:
|        Fresh unicode-xid v0.2.0
|        Fresh cc v1.0.50
|        Fresh autocfg v1.0.0
|        Fresh cfg-if v0.1.10
|        Fresh version_check v0.1.5
|        Fresh either v1.5.3
|        Fresh glob v0.3.0
|        Fresh regex-syntax v0.6.14
|        Fresh lazy_static v1.4.0
|        Fresh bytes v0.5.4
|        Fresh pkg-config v0.3.17
|        Fresh toml v0.2.1
|        Fresh error-chain v0.10.0
|        Fresh termcolor v1.1.0
|        Fresh unicode-segmentation v1.6.0
|        Fresh rustc-hash v1.1.0
|        Fresh fixedbitset v0.2.0
|        Fresh peeking_take_while v0.1.2
|        Fresh shlex v0.1.1
|        Fresh remove_dir_all v0.5.2
|        Fresh lazycell v1.2.1
|        Fresh futures-core v0.3.4
|        Fresh multimap v0.8.0
|        Fresh pin-project-lite v0.1.4
|        Fresh fnv v1.0.6
|        Fresh itoa v0.4.5
|        Fresh ppv-lite86 v0.2.6
|        Fresh pin-utils v0.1.0-alpha.4
|        Fresh tower-service v0.3.0
|        Fresh futures-sink v0.3.4
|        Fresh futures-task v0.3.4
|        Fresh unicode-width v0.1.7
|        Fresh percent-encoding v1.0.1
|        Fresh void v1.0.2
|        Fresh vec_map v0.8.1
|        Fresh strsim v0.8.0
|        Fresh ansi_term v0.11.0
|        Fresh quick-error v1.2.3
|        Fresh arc-swap v0.4.5
|        Fresh hex v0.4.2
|        Fresh cmake v0.1.42
|        Fresh thread_local v1.0.1
|        Fresh itertools v0.8.2
|        Fresh tracing-core v0.1.9
|        Fresh metadeps v1.1.2
|        Fresh heck v0.3.1
|        Fresh http v0.2.0
|        Fresh tokio v0.2.11
|        Fresh c2-chacha v0.2.3
[... 46 lines cut ...]
|   DEBUG = Some("true")
|   CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
|   running: "/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/wrappers/cc-native-wrapper.sh" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-O2" "-pipe" "-g" "-feliminate-unused-debug-types" "-fmacro-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot=" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot-native=" "-o" "/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/target/release/build/libloading-8b4570a569a949e1/out/src/os/unix/global_static.o" "-c" "src/os/unix/global_static.c"
|   cargo:warning=gcc: error: unrecognized command line option ‘-fmacro-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1’
|   exit code: 1
| 
|   --- stderr
| 
| 
|   error occurred: Command "/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/wrappers/cc-native-wrapper.sh" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-O2" "-pipe" "-g" "-feliminate-unused-debug-types" "-fmacro-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/rasp [... cut ...]
ERROR: Task (/Projects/chirpstack-gateway-os/layers/chirpstack/meta-chirpstack/recipes-chirpstack/chirpstack-concentratord/chirpstack-concentratord_3.0.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 642 tasks of which 641 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /Projects/chirpstack-gateway-os/layers/chirpstack/meta-chirpstack/recipes-chirpstack/chirpstack-concentratord/chirpstack-concentratord_3.0.2.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

Thanks

SLDP-8255, 2020-12-15:
@brocaar 
Any advice would be appreciated,
Thanks

brocaar, 2020-12-21:
Are you using the provided Docker Compose environment?

SLDP-8255, 2021-01-05:
@brocaar Thanks for your interest.
Yes, the problem occurs both during build with use of docker-compose as well in the case of direct build in the host OS (without container).
In addition, I changed and set `PARALLEL_MAKE` variable to `6`, because with its default value (6*2 in my case) I was experiencing the [out of memory issue while compiling clang](https://github.com/brocaar/chirpstack-gateway-os/issues/60).

brocaar, 2021-01-06:
Looking at the logs, it seems these are not from the provided Compose environment (the Compose env. is not using `/Projects/...` as prefix): 

> running: "/Projects/chirpstack-gateway-os/build/tmp/raspberrypi/raspberrypi3-glibc/...

Could you post the logs when using the provided Compose environment?

SLDP-8255, 2021-01-07:
@brocaar Yes, the logs I attached in the first post are about building directly in the host system (Ubuntu 18.04).
Sorry, I forgot to add Compose environment logs in the last message. They are as follows:
```
|   error occurred: Command "/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/wrappers/cc-native-wrapper.sh" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-O2" "-pipe" "-g" "-feliminate-unused-debug-types" "-fmacro-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot=" "-fdebug-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot-native=" "-o" "/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/target/release/build/libloading-8b4570a569a949e1/out/src/os/unix/global_static.o" "-c" "src/os/unix/global_static.c" with args "cc-native-wrapper.sh" did not execute successfully (status code exit code: 1).
| 
| 
| warning: build failed, waiting for other jobs to finish...
| error: build failed
| WARNING: exit code 101 from a shell command.
| ERROR: Execution of '/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/temp/run.do_compile.112' failed with exit code 101:
|        Fresh unicode-xid v0.2.0
|        Fresh cc v1.0.50
|        Fresh autocfg v1.0.0
|        Fresh cfg-if v0.1.10
|        Fresh version_check v0.1.5
|        Fresh either v1.5.3
|        Fresh glob v0.3.0
|        Fresh lazy_static v1.4.0
|        Fresh regex-syntax v0.6.14
|        Fresh error-chain v0.10.0
|        Fresh toml v0.2.1
|        Fresh pkg-config v0.3.17
|        Fresh termcolor v1.1.0
|        Fresh bytes v0.5.4
|        Fresh shlex v0.1.1
|        Fresh remove_dir_all v0.5.2
|        Fresh lazycell v1.2.1
|        Fresh rustc-hash v1.1.0
|        Fresh unicode-segmentation v1.6.0
|        Fresh fixedbitset v0.2.0
|        Fresh peeking_take_while v0.1.2
|        Fresh futures-core v0.3.4
|        Fresh pin-project-lite v0.1.4
|        Fresh multimap v0.8.0
|        Fresh itoa v0.4.5
|        Fresh ppv-lite86 v0.2.6
|        Fresh fnv v1.0.6
|        Fresh futures-sink v0.3.4
|        Fresh futures-task v0.3.4
|        Fresh pin-utils v0.1.0-alpha.4
|        Fresh tower-service v0.3.0
|        Fresh void v1.0.2
|        Fresh unicode-width v0.1.7
|        Fresh percent-encoding v1.0.1
|        Fresh strsim v0.8.0
|        Fresh quick-error v1.2.3
|        Fresh vec_map v0.8.1
|        Fresh ansi_term v0.11.0
|        Fresh arc-swap v0.4.5
|        Fresh hex v0.4.2
|        Fresh cmake v0.1.42
|        Fresh itertools v0.8.2
|        Fresh thread_local v1.0.1
|        Fresh tracing-core v0.1.9
|        Fresh metadeps v1.1.2
|        Fresh heck v0.3.1
|        Fresh tokio v0.2.11
|        Fresh http v0.2.0
|        Fresh c2-chacha v0.2.3
|        Fresh futures-util v0.3.4
|        Fresh textwrap v0.11.0
|        Fresh humantime v1.3.0
|        Fresh proc-macro2 v1.0.8
|        Fresh libc v0.2.66
|    Compiling libloading v0.5.2
|        Fresh log v0.4.8
|        Fresh zeromq-src v0.1.8+4.3.2
|      Running `/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/target/release/build/libloading-cb7b018aa809cc7e/build-script-build`
|        Fresh bitflags v1.2.1
|        Fresh memchr v2.3.2
[... 35 lines cut ...]
|   CFLAGS = Some(" -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1                      -fdebug-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1                      -fdebug-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot=                      -fdebug-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/recipe-sysroot-native= ")
|   CRATE_CC_NO_DEFAULTS = None
|   DEBUG = Some("true")
|   CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
|   running: "/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1/wrappers/cc-native-wrapper.sh" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-O2" "-pipe" "-g" "-feliminate-unused-debug-types" "-fmacro-prefix-map=/build/tmp/raspberrypi/raspberrypi3-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chirpstack-concentratord/3.0.2-r1=/usr/src/debug/chirpstack-concentratord/3.0.2-r1" "-fdebug-pr [... cut ...]
ERROR: Task (/chirpstack-gateway-os/layers/chirpstack/meta-chirpstack/recipes-chirpstack/chirpstack-concentratord/chirpstack-concentratord_3.0.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 642 tasks of which 641 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /chirpstack-gateway-os/layers/chirpstack/meta-chirpstack/recipes-chirpstack/chirpstack-concentratord/chirpstack-concentratord_3.0.2.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

SLDP-8255, 2022-08-18:
That was caused by the docker container that was not rebuilt and updated.

closed 2022-08-18 by SLDP-8255