chirpstack/chirpstack-gateway-os #72: Concentratord Problem with Risinghf hat on Raspberry Pi 3

olman011, 2021-01-28:
I’m using the full OS installation to try to get this system running.
I have a RPI 3 with a RHF0M301 hat.

When I go through the setup for the concentrator shield and select option 9 Risinghf RHF0M301 and select US915 and the first channel option and then yes to all the file write options, I am returned to the main menu.

At the top it says
Version 3.3.3
Gateway ID: not configured

When I look at the general configuration for Concentratord it shows
model=“generic_us915”

and at the end of the file it shows
reset_pin=0

When I exit to the command line and enter sudo monit summary it tells me everything is OK except the chirpstack-concentratord which has the status of “does not exist”.

freeedom67, 2021-02-16:
Hello, I have exact the same problem with the same hardware and configuration. I checked everything and cant find the reason for this. There is also nopc error message or somtething to see in the logs except a permanent restart of the concentratord service.

freeedom67, 2021-02-16:
When i try to start the chirpstack-concentratord service alone I get at least an error message:

ERROR: Failed to load fw 1
ERROR:Version of callibration firmware not expected, acutal=0 expected:2

Thread 'main' panicked at called Result::unwrap() on an 'Err' value: 'lgw_start failed'

freeedom67, 2021-02-18:
I think this issue is realated to : https://github.com/brocaar/chirpstack-gateway-os/issues/71

I found out, that the RHF0M301 board needs the reset pin 7, and this one is occupied by the system so i get the desribed SYS(Busy) error when starting the concentratord. Unfortunatly the fix in https://github.com/brocaar/chirpstack-gateway-os/issues/71 doesnt work for me.

elsni, 2021-09-09:
is there ANY workaround?

brocaar, 2021-09-13:
Yes, I'll include a fix which includes:

- A Concentratord profile for the RisingHF shield
- Device-tree overlay config in the Gateway OS to make GPIO7 usable

I already have a working Raspberry Pi + RisingHF shield locally :) I'll push my changes tomorrow after some more testing.

brocaar, 2021-09-15:
The latest ChirpStack Concentratord contains configuration for the RisingHF shield (EU868 + US915). 

https://github.com/brocaar/chirpstack-gateway-os/commit/9090ce304dd1a6a0b8a6b1f7bce17bd94dff858b includes an overlay to "free" GPIO7 and https://github.com/brocaar/chirpstack-gateway-os/commit/82e46e6b1897cdd71f577c61990630ce7633f0cf contains `gateway-config` updates to properly configure the RisingHF shield.

closed 2021-09-15 by brocaar