chirpstack/chirpstack-gateway-os #74: IN865 on Gateway OS base

saiprakash1407, 2021-04-15:
Gateway OS base image does not contains IN865 for RAK2245,
is there any other method through which i can configure it?

manivalaguru, 2021-04-21:
have to got it working in IN865? i am also doing the same thing

saiprakash1407, 2021-04-29:
> have to got it working in IN865? 

NO

manivalaguru, 2021-04-29:
okay, i have done that

saiprakash1407, 2021-04-29:
> okay, i have done that

Can please tell me what changes we need to do.

manivalaguru, 2021-04-29:
no bro i have used lora packet forwarder chirpstack bridge in my gateway and application and netwrk server sepretely

saiprakash1407, 2021-04-29:
your not using chirpstack gateway os base.?

> no bro i have used lora packet forwarder chirpstack bridge in my gateway and application and netwrk server sepretely
you are  not using chirpstack gateway os base.?

manivalaguru, 2021-04-29:
no using only chirpstack servers and bridge

brocaar, 2021-09-15:
This has been added in https://github.com/brocaar/chirpstack-gateway-os/commit/82e46e6b1897cdd71f577c61990630ce7633f0cf.

closed 2021-09-15 by brocaar