chirpstack/chirpstack-gateway-os #75: Concentratord hangs - restart required(RPI3+RAK2245)

saiprakash1407, 2021-04-15:
Hi,

I am running Chirpstack Base OS (3.3.3) on RPi3 with RAK2245.

The Chirpstack GW runs good. And after a certain period the gateway goes to offline.

However when i resetting the concentrator  the problem is solved and gateway coming to online. This only lasts for a certain time and reset is required again.

sudo monit restart chirpstack-concentratord

[image]

brocaar, 2021-04-27:
Couple of questions:

> And after a certain period the gateway goes to offline.

Could you define "certain time"? is that minutes, hours, days. ...?

The logs that you posted cover less than a minute. Is this the point where the Concentratord hangs? To understand what is happening I would like to ask you to post the logs with `TRACE` level. Instead of posting a screenshot, please share the logs as a file.

saiprakash1407, 2021-04-28:
Hi brocaar,

Within one minute gateway going to offline.

[concentratord.txt](https://github.com/brocaar/chirpstack-gateway-os/files/6390953/concentratord.txt)

[gateway-bridge.txt](https://github.com/brocaar/chirpstack-gateway-os/files/6390963/gateway-bridge.txt)

brocaar, 2021-04-28:
That sounds like the issue is because of the config command sent to the gateway. Could you try removing the gateway-profile in the gateway configuration (web-interface) and restart the Concentratord? My assumption is that the configuration that you're sending to the gateway is invalid and causes the gateway to no longer work.

I'm running the same combination of hardware and versions for ~ 1 day now without issues.

saiprakash1407, 2021-04-29:
Hi brocaar,

I removed gateway-profile in the gateway configuration. now gateway working fine. thanks for your support.

Can you please help me how to configure IN865 in gateway os.

closed 2021-09-07 by brocaar