chirpstack/chirpstack-gateway-os #76: issue changing channel configuration to in 866 in chripstack os

manivalaguru, 2021-04-19:
hello,
i am using ic880a with chripstack os and raspberry pi by default my channel configuration is 868 i want to change it to 866 my understanding is i want to do the changes in global_config.json but i cant be able to find the file can you help me understand this
thanks and regards
manikandan v

saiprakash1407, 2021-04-29:
Hi manikandan,

I am also facing the same problem. did you get the any solution for this?

iyer3142, 2021-08-12:
I looked at all the files in /opt, /etc folders and changed files for EU868 to match the frequencies in India. I got my node communicating with the gateway sucessfully. I would recommend that you do the same :)

brocaar, 2021-09-15:
This has been implemented in the above commit + https://github.com/brocaar/chirpstack-gateway-os/commit/82e46e6b1897cdd71f577c61990630ce7633f0cf.

closed 2021-09-15 by brocaar

iyer3142, 2021-09-16:
Can you confirm if the IN865 has been implemented for SX1301 & 1302 too?

brocaar, 2021-09-20:
I focused on this issue, which was IN865 for the iC880a concentrator shield (SX1301 chipset). Please open an other issue or pull-request for different questions :)