chirpstack/chirpstack-gateway-os #77: Acivating a Dragino LHT65

landersson936, 2021-07-02:
I have the Gateway OS installed on a Pi3B+ and and all services are functioning. (No internet connection)
I have set up service profile, device profile and an application.
I get error:"join-server returned error: response error, code: MICFailed, description: invalid mic"
when trying to activate a LHT65.

For the network key and Application key, I used the info from the device sticker called NETSKEY and APPSKEY .

Is that correct?

landersson936, 2021-07-02:
The LHT65 firmware is  V1.7 and has LoRaWan stack dr-lws-004

brocaar, 2021-09-14:
> "join-server returned error: response error, code: MICFailed, description: invalid mic"

This issue is related to incorrect root-keys and / or configuration. Please search for this error message on the forum as it has been discussed a couple of times: https://forum.chirpstack.io/.

closed 2021-09-14 by brocaar