chirpstack/chirpstack-gateway-os #87: Build error - chirpstack_4.0.0.-rc.2

SLDP-8255, 2022-08-18:
- [ ] The issue is present in the latest release.
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?
During Yocto build process, compilation errors occured for package: chirpstack_4.0.0-rc.2.
I cloned fresh repository and set revision to 4.0.0-rc.1 and MACHINE variable to "raspberrypi4-64" (but I think it shouldn't affect compilation of this package).  Then, a fresh docker environment has created and Yocto building process started.

## What did you expect?
No compilation errors for this package.
## Steps to reproduce this issue

Steps:

1. Clone repository
2. make submodules
3. make permissions
4. set MACHINE to "raspberrypi4-64" in meta/conf/local.conf.sample
5. Launch build in the way as shown in the readme.md 

## Could you share your log output?
```
Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 14 Local 0 Mirrors 0 Missed 14 Current 1656 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: chirpstack-4.0.0-rc.2-r0 do_compile: ExecutionError('/build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/temp/run.do_compile.105789', 101, None, None)
ERROR: Logfile of failure stored in: /build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/temp/log.do_compile.105789
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 32
| yarn install
| yarn install v1.22.18
| warning Skipping preferred cache folder "/home/yocto/.cache/yarn" because it is not writable.
| warning Selected the next writable cache folder in the list, will be "/tmp/.yarn-cache-999".
| [1/4] Resolving packages...
| [2/4] Fetching packages...
| [3/4] Linking dependencies...
| warning " > @testing-library/user-event@12.8.3" has unmet peer dependency "@testing-library/dom@>=7.21.4".
| warning " > chartjs-adapter-moment@1.0.0" has unmet peer dependency "moment@^2.10.2".
| warning " > react-codemirror2@7.2.1" has incorrect peer dependency "react@>=15.5 <=16.x".
| warning "react-scripts > eslint-config-react-app > eslint-plugin-flowtype@8.0.3" has unmet peer dependency "@babel/plugin-syntax-flow@^7.14.5".
| warning "react-scripts > eslint-config-react-app > eslint-plugin-flowtype@8.0.3" has unmet peer dependency "@babel/plugin-transform-react-jsx@^7.14.9".
| [4/4] Building fresh packages...
| success Saved lockfile.
| Done in 6.37s.
| yarn add file:../api/grpc-web
| yarn add v1.22.18
| warning Skipping preferred cache folder "/home/yocto/.cache/yarn" because it is not writable.
| warning Selected the next writable cache folder in the list, will be "/tmp/.yarn-cache-999".
| [1/4] Resolving packages...
| [2/4] Fetching packages...
| [3/4] Linking dependencies...
| warning " > @testing-library/user-event@12.8.3" has unmet peer dependency "@testing-library/dom@>=7.21.4".
| warning " > chartjs-adapter-moment@1.0.0" has unmet peer dependency "moment@^2.10.2".
| warning " > react-codemirror2@7.2.1" has incorrect peer dependency "react@>=15.5 <=16.x".
| warning "react-scripts > eslint-config-react-app > eslint-plugin-flowtype@8.0.3" has unmet peer dependency "@babel/plugin-syntax-flow@^7.14.5".
| warning "react-scripts > eslint-config-react-app > eslint-plugin-flowtype@8.0.3" has unmet peer dependency "@babel/plugin-transform-react-jsx@^7.14.9".
| [4/4] Building fresh packages...
| success Saved lockfile.
| success Saved 1 new dependency.
| info Direct dependencies
| └─ @chirpstack/chirpstack-api-grpc-web@4.0.0-rc.2
| info All dependencies
| └─ @chirpstack/chirpstack-api-grpc-web@4.0.0-rc.2
| Done in 6.65s.
| yarn build
| yarn run v1.22.18
| warning Skipping preferred cache folder "/home/yocto/.cache/yarn" because it is not writable.
| warning Selected the next writable cache folder in the list, will be "/tmp/.yarn-cache-999".
| $ react-scripts build
| warning Cannot find a suitable global folder. Tried these: "/usr/local, /home/yocto/.yarn"
[... 513 lines cut ...]
| 
| Caused by:
|   process didn't exit successfully: `rustc --crate-name rquickjs_sys --edition=2018 /build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/cargo_home/bitbake/rquickjs-sys-0.1.6/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=z -C linker-plugin-lto -C codegen-units=1 --cfg 'feature="bindgen"' --cfg 'feature="bindgen-rs"' --cfg 'feature="exports"' -C metadata=c559bcdb10e8e735 -C extra-filename=-c559bcdb10e8e735 --out-dir /build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/build/target/aarch64-oe-linux/release/deps --target aarch64-oe-linux -C linker=/build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/wrapper/target-rust-ccld -L dependency=/build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/build/target/aarch64-oe-linux/release/deps -L dependency=/build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/build/target/release/deps --cap-lints allow -L /build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0=/usr/src/debug/chirpstack/4.0.0-rc.2-r0 -L native=/build/tmp/raspberrypi4-64-glibc/work/cortexa72-oe-linux/chirpstack/4.0.0-rc.2-r0/build/target/aarch64-oe-linux/release/build/rquickjs-sys-f0d5f8a4503cfd99/out -l static=quickjs` (exit status: 1)
| warning: build failed, waiting for other jobs to finish...
|     Building [=======================> ] 545/550: diesel_migrations, lrwn
| warning: `diesel_migrations` (lib) generated 2 warnings (2 duplicates)
|     Building [=======================> ] 546/550: lrwn
| warning: `lrwn` (lib) generated 2 warnings (2 duplicates)
| error: build failed
| WARNING: exit code 101 from a shell command.
ERROR: Task (/chirpstack-gateway-os/meta/recipes-chirpstack/chirpstack/chirpstack_4.0.0-rc.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4025 tasks of which 4023 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /chirpstack-gateway-os/meta/recipes-chirpstack/chirpstack/chirpstack_4.0.0-rc.2.bb:do_compile
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

## Your Environment

Host OS: Ubuntu 22.04

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | v4.0.0-rc.1  |

Patrick10281988, 2023-11-13:
@SLDP-8255 , have you found the solution to this issue. currently i am facing the exactly the same.

brocaar, 2023-12-22:
I'm closing this issue, as it is no longer relevant after the OpenWrt migration.

Hint: I believe this issue was caused by an incorrect bindgen target. E.g. it generated code for the host platform, not the target platform (which would result in different code being generated). You can set this flag (with probably a different value, as this is for OpenWrt): https://github.com/chirpstack/chirpstack-openwrt-feed/blob/master/chirpstack/chirpstack/Makefile#L21

closed 2023-12-22 by brocaar