chirpstack/chirpstack-gateway-os #89: LoRaGo PORT not working

mjohanning99, 2022-09-19:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?
I have connected a LoRaGo PORT to my Raspberry Pi 3 and used the LoRaGo PORT configuration in the "Setup LoRa concentrator shield" menu in gateway-config. However, no lights are glowing and no gateway ID is being configured. 

I did see a closed issue here (#66) and commented on it a while ago, but there has not yet been an answer. As it still doesn't work months later, I decided to open another issue. 

## What did you expect?
The concentrator to start and work and to be given a gateway ID. 

## Steps to reproduce this issue
Connect the LoRaGo PORT to a Raspberry Pi and configuring it using gateway-config.

## Could you share your log output?

```shell
ting up concentrator channels
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring multi-SF LoRa channel, channel: 0, enabled: true, freq: 868100000, rf_chain: 1, if_freq: -400000
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring multi-SF LoRa channel, channel: 1, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 868500000, rf_chain: 1, if_freq: 0
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 867100000, rf_chain: 0, if_freq: -400000
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring multi-SF LoRa channel, channel: 4, enabled: true, freq: 867300000, rf_chain: 0, if_freq: -200000
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring multi-SF LoRa channel, channel: 5, enabled: true, freq: 867500000, rf_chain: 0, if_freq: 0
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring multi-SF LoRa channel, channel: 6, enabled: true, freq: 867700000, rf_chain: 0, if_freq: 200000
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 867900000, rf_chain: 0, if_freq: 400000
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring Std LoRa channel, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Configuring FSK channel, enabled: true, freq: 868800000, rf_chain: 1, if_freq: 300000
Sep 19 09:03:58 raspberrypi3 user.info chirpstack-concentratord-sx1301[942]: Starting the concentrator
Sep 19 09:04:00 raspberrypi3 user.err chirpstack-gateway-bridge[592]: time="2022-09-19T09:04:00.913589202Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial un
Sep 19 09:04:04 raspberrypi3 user.err chirpstack-gateway-bridge[592]: time="2022-09-19T09:04:04.427206744Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial un
Sep 19 09:04:07 raspberrypi3 user.err chirpstack-gateway-bridge[592]: time="2022-09-19T09:04:07.940586578Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial un
Sep 19 09:04:08 raspberrypi3 user.err monit[430]: 'chirpstack-concentratord' process is not running
Sep 19 09:04:08 raspberrypi3 user.info monit[430]: 'chirpstack-concentratord' trying to restart
Sep 19 09:04:08 raspberrypi3 user.info monit[430]: 'chirpstack-concentratord' start: '/etc/init.d/chirpstack-concentratord start'
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring reset pin, chip: 0, pin: 0
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Starting Concentratord SX1301 (version: 3.3.2, docs: https://www.chirpstack.io/concentratord/)
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Triggering concentrator reset
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Setting spi device path, spidev_path: /dev/spidev0.0
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Setting board configuration, lorawan_public: true, clock_source: 1
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Setting up concentrator radios
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring radio, radio: 0, enabled: true, center_freq: 867500000, type: SX1257
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring radio, radio: 1, enabled: true, center_freq: 868500000, type: SX1257
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Setting up concentrator channels
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring multi-SF LoRa channel, channel: 0, enabled: true, freq: 868100000, rf_chain: 1, if_freq: -400000
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring multi-SF LoRa channel, channel: 1, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 868500000, rf_chain: 1, if_freq: 0
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 867100000, rf_chain: 0, if_freq: -400000
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring multi-SF LoRa channel, channel: 4, enabled: true, freq: 867300000, rf_chain: 0, if_freq: -200000
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring multi-SF LoRa channel, channel: 5, enabled: true, freq: 867500000, rf_chain: 0, if_freq: 0
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring multi-SF LoRa channel, channel: 6, enabled: true, freq: 867700000, rf_chain: 0, if_freq: 200000
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 867900000, rf_chain: 0, if_freq: 400000
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring Std LoRa channel, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Configuring FSK channel, enabled: true, freq: 868800000, rf_chain: 1, if_freq: 300000
Sep 19 09:04:08 raspberrypi3 user.info chirpstack-concentratord-sx1301[949]: Starting the concentrator
Sep 19 09:04:11 raspberrypi3 user.err chirpstack-gateway-bridge[592]: time="2022-09-19T09:04:11.453818289Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial un
Sep 19 09:04:14 raspberrypi3 user.err chirpstack-gateway-bridge[592]: time="2022-09-19T09:04:14.966808536Z" level=error msg="backend/concentratord: event socket dial error" error="dial event api url error: zmq4: could not dial to \"ipc:///tmp/concentratord_event\": dial un

```

## Your Environment
Gateway OS 3.6.0 base 
Raspberry Pi 3 

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | 3.6.0 base  |
| Raspberry Pi | 3 |

mjohanning99, 2022-09-19:
So after trying some more stuff, what finally worked was using the configuration for the iC880A and setting the reset pin to be 25. Not entirely sure why the default configuration for the LoRaGo PORT doesn't work; maybe something needs to be changed in the configuration for it?

mjohanning99, 2022-09-20:
And now it doesn't work anymore. Even when doing the same thing as yesterday, the same error I had has come back.

mjohanning99, 2022-09-21:
I decided to upgrade to version 4.0.0 and still could not get the LoRaPort GO to work. However, by selecting the iC880A gateway configuration in `gateway-config` and then manually setting the `reset_pin` in `/etc/chirpstack-concentratord/sx1301/concentratord.toml` and _then_ restarting concentratord through `gateway-config`, I was able to consistently get the LoRaPort GO shields I have working. Setting the reset pin through the TUI of `gateway-config` and does not anything. It worked only after manually setting the value of `reset_pin` in the above-mentioned configuration file to 25

It really seems like there is a misconfigured reset pin somewhere in the default config for the LoRaPort GO; though that would not explain how setting a reset pin through `gateway-config` did not work.

brocaar, 2023-12-22:
Is this still an issue with the latest OpenWrt based ChirpStack Gateway OS?

closed 2024-02-02 by brocaar