chirpstack/chirpstack-gateway-os #95: NODE RED problem on version 4.0.1

stathisioannides, 2022-12-01:
We were using NODE RED on version 4.0 with no problems.

We updated to version 4.0.1 using : chirpstack-gateway-os-full-update-raspberrypi4-20221114095641.swu
but after reboot of Pi4 was impossible to connect on NODE RED ( IP:1880 ) althought it was enabled.

We disabled and reenabled it through gateway-config,  after that we had a succesfull connection but the flows we had built where changed. We build them again and the system start working.

But after 2-3 message transfers connection with NODE RED server was lost again. We tried the disable-enable cycle of NODE RED but the problem persists.

brocaar, 2022-12-06:
What are the minimum steps to reproduce this issue? Could you further explain what this means?

> But after 2-3 message transfers connection with NODE RED server was lost again. 

What does "message transfers" mean in this case?

brocaar, 2022-12-12:
Thanks, I'm able to reproduce the issue. For some reason, NodeJS is throwing a "Bus error" and terminates.

```
12 Dec 12:23:22 - [info] Starting flows
12 Dec 12:23:22 - [info] Started flows
12 Dec 12:23:22 - [info] [mqtt-broker:localhost] Connected to broker: mqtt://localhost:1883
Bus error
```

I'll investigate this further.

brocaar, 2022-12-13:
Quick update, I've tried various things:

* Updating Node-RED
* Updating node-red-contrib-chirpstack package
* Different ways of packaging the above packages
* Installing these packages by hand on the Pi

But this doesn't solve the issue. I'm now trying to update NodeJS to a later version (16.18.1) in the hope that this will resolve the issue :crossed_fingers: 

There have been several reports by other uses (non ChirpStack related), in which case it was either resolved by deleting the `node_packages` directory and `npm install ...` everything again or updating NodeJS.

brocaar, 2022-12-15:
@stathisioannides the above commit fixes this issue.

commit 6cb9171b (2022-12-15): Revert "Update to langdale."

This reverts commit 669d130191db0ce0d2d2e33162307834ade5e559.

This update caused Node-RED / NodeJS to fail with "Bus error".

Closes #95.
closed 2022-12-15 by brocaar