chirpstack/chirpstack-gateway-os #96: ChirpStack Gateway OS v4.0.1 doesn't boot on Raspberry PI 4B V1.5 revision do3115 Soc BCM2711, 8GB RAM

dyane6, 2022-12-05:
Hardware:
- Raspberry PI 4B V1.5 revision do3115 Soc BCM2711, 8GB RAM
- Waveshare SX1302 hat  

Software:
Gateway OS V4.0.1

- [X] The issue is present in the latest release.
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## Issue
- the boot fails and the screen is frozen after the following strings
- the board is not seen on the LAN

## Steps to reproduce this issue
Steps:
1. Flash the last OS Gateway software on SD card 32 Gb
2. Power on the Raspberry PI 4B
3. The board is Ethernet wired on the LAN 

## log output:
- the boot fails and the screen is frozen after the following strings
- Here are the strings displayed on the screen:
      Net:     eth0:    ethernet@7d580000
      PCIe BRCM: link up, 5.0 Gbps x 1 (SSC)
      PCI: Failed autoconfig bar 10
      starting USB...
      Bus xhci_pci:

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | v4.0.1  |

dloftus98, 2022-12-08:
same error for me... can't swupdate to it or install it fresh.  Mine is a rpi 4b with rak2245.

brocaar, 2023-12-22:
I'm closing this issue as I'm not sure if it is still relevant after Yocto -> OpenWrt migration.

closed 2023-12-22 by brocaar