chirpstack/chirpstack-gateway-os #98: IN865 configuration  for SX1302?

dev-techshlok, 2023-05-22:
## Summary

We have been working with separate SX1302 binaries and chirpstack server for last a few months, it works well.
But is there any way to integrate IN865 in SX1302 chipset configuration for corecell and waveshare gateway profile, for gateway-os

## What is the use-case?

Chirpstack gateway os with IN865 configurations

Please guide me.

dev-techshlok, 2023-05-23:
Tried adding the config files as 
 `sudo /usr/bin/chirpstack-concentratord-sx1302 -c /etc/chirpstack-concentratord/sx1302/concentratord.toml -c /etc/chirpstack-concentratord/sx1302/band.toml -c /etc/chirpstack-concentratord/sx1302/channels.tomltoml -c /etc/chirpstack-concentratord/sx1302/band.toml`
 
 
 But for band.toml it is empty for in865 and channel also only have half info 
 
```
 # LoRa concentrator configuration.
[gateway.concentrator]

# Multi spreading-factor channels (LoRa).
multi_sf_channels=[
  865062500,
  865402500,
  865985000,
  0,
  0,
  0,
  0,
  0,]

```

This seems to not start.

brocaar, 2024-03-28:
* Waveshare already has IN865 config: https://github.com/chirpstack/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1302/src/config/vendor/waveshare/sx1302_lorawan_gateway_hat.rs#L17
* I have just pushed two commits to add this to the CoreCell configs:
    * https://github.com/chirpstack/chirpstack-concentratord/commit/915c6821cd9cb61e3e050b1bf4b139bea456ec6b
    * https://github.com/chirpstack/chirpstack-concentratord/commit/3c7ba7f2863742a116af2b0b9e1c530f75c48e1c

Once a new Concentratord version has been released, this needs to be added to the LuCI config: https://github.com/chirpstack/chirpstack-openwrt-feed/blob/master/chirpstack/luci-app-chirpstack-concentratord-target-rpi/htdocs/luci-static/resources/view/chirpstack/concentratord.js