chirpstack/chirpstack-gateway-os #101: 4.2.0 Missing device profile templates

scottmil, 2023-09-14:
When I installed ChirpStack OS 4.2.0, there were no device profile templates in the drop-down when I attempted to add devices to an application.

Raspberry Pi 4 Model B
RAK7248 Gateway

boba2z, 2024-01-05:
Same issue here

Raspberry Pi 4 Model B

TerraControlS, 2024-03-17:
@brocaar - Also the same issue, using image chirpstack-gateway-os-4.3.2-full-bcm27xx-bcm2709-rpi-2-squashfs-factory.img

I would just like to add, after spending the last week navigating Chirpstack and various forum posts, you have created an awesome product and your ability to support users is impressive! Seems like quite a thankless endeavor so I thought I would show some appreciation.

brocaar, 2024-03-28:
This has been added back in the above commits.

closed 2024-03-28 by brocaar