chirpstack/chirpstack-gateway-os #104: Waveshare sx1302 Raspberry HAT US915 frequency plan selection invalid

jcarre59, 2023-12-12:
- The issue is present in the latest release installation Openwrt Image for rasberry pi
- I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?

When I select Waveshare sx1302 Raspberry HAT  all frequency plans selection turn red invalid when I try to use US915

## Steps to reproduce this issue

Steps:

1. I reinstalled from scratch and still the same.

I use download image from official chirpstack site 

uname -a

Linux chirpstack-3f454f 5.15.127 #0 SMP Fri Sep 1 13:17:30 2023 armv7l GNU/Linux
[image]

jcarre59, 2023-12-12:
If I run it manually:

root@chirpstack-3f454f:/var/etc/chirpstack-concentratord# /usr/bin/chirpstack-concentratord-sx1302 -c /var/etc/chirpstack-concentratord/concentratord.toml
-c /var/etc/chirpstack-concentratord/region.toml -c /var/etc/chirpstack-concentratord/channels.toml
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Region is not supported: US915', chirpstack-concentratord-sx1302/src/config/mod.rs:221:73
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

brocaar, 2024-02-02:
This was already fixed in the previous version I believe. There will also be a new GW OS update with some more improvements.

closed 2024-02-02 by brocaar