chirpstack/chirpstack-gateway-os #105: USB Serial with Gateway OS on Rpi

STG-Jamie, 2023-12-20:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?
Unable to install kmod-usb-serial on Rpi4 running the latest Gateway OS (4.2.0)

## What did you expect?
The currently available package is for kmod_usb_serial is 5.15.139 however as the kernel is 5.15.127 it is deemed incompatible. Either update kernel or make a 5.15.127 version of kmod-usb-serial available

## Steps to reproduce this issue

Steps:

1. Try installing kmod-usb_serial package
2.
3.
4.

## Could you share your log output?

```shell

```

## Your Environment

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | v4.2.0  |

brocaar, 2023-12-22:
Could you share your use-case + describe what kind of device you would like to connect? Does this also require the `ftdi` kernel mod?

[image]

STG-Jamie, 2023-12-22:
We have a non-lora radio module that acts as a hub for a star network of wireless sensors whos’ data we want to access in the gateway. 
The radio module interface is a TTL level UART.
I intend to use an FTDI USB-Serial converter.
Generic virtual com port operation is all we need e.g. set up standard config e.g. baud rate so not sure if we need the ftdi specific kernel mod as well.

brocaar, 2024-02-01:
I'm now testing this and can confirm that in the current version, a FTDI device is not detected. I'm now testing to see if `kmod-usb-serial` alone would be enough, I assume we actually need `kmod-usb-serial-ftdi`.

commit 1bc9ff0c (2024-02-01): rpi: Add kmod-usb-serial-ftdi.

Fixes #105.
closed 2024-02-01 by brocaar