chirpstack/chirpstack-gateway-os #106: RAK7268v2 support

Arachnid, 2024-01-06:
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## Summary

I see recent commits mentioning support for running on RAK7268v2. Is there a timeline for a release supporting this hardware, or any docs on how to try it out pre-release?

KenyOS, 2024-04-25:
I followed the doc on [https://www.chirpstack.io/docs/chirpstack-mqtt-forwarder/install/rak.html](https://www.chirpstack.io/docs/chirpstack-mqtt-forwarder/install/rak.html)

on RAK7268V2
I get an error when trying to install.

```
root@RAK7268V2:/tmp# opkg install chirpstack-mqtt-forwarder_4.2.3-r1_mipsel_24kc.ipk
Unknown package 'chirpstack-mqtt-forwarder'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for chirpstack-mqtt-forwarder found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package chirpstack-mqtt-forwarder.
```

brocaar, 2024-10-28:
RAK72682v2 images are now provided: https://www.chirpstack.io/docs/chirpstack-gateway-os/install/rak.html

closed 2024-10-28 by brocaar