chirpstack/chirpstack-gateway-os #107: CS OS 4.3.0 doesn't work on RPi 4B board

marekpow, 2024-01-13:
- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

My infrastructure:
[image]
Fresh Chirpstack 4.3.0 is installed on this board.

My router shows CS attached to the network:
[image]

And this is the same as on the CS OS console:
[image]

## What happened?
CS OS 4.3.0 doesn't work on the RPi 4B. 
Ping doesn't respond
[image]

The Web interface doesn't respond:
[image]

## What did you expect?
Should work from without any problems.
## Steps to reproduce this issue

Steps:

1.Get RPi 4B
2.Prepare SD disk with CS OS verison 4.3.0 (FULL)
3.Insert SD disk  into RPi 4 board
4.Power on
5. Try to open web interface to CS OS OPenWrt (in my cases was the IP address 192.168.0.181)
6. Ping to this address

## Could you share your log output?

```shell

It doesn't work. Answer:
tail: can't open /var/log/messages : no such file or directory
tail: no files

I have got from the log a few screenshots with suspicious messages 

```

## Your Environment
[image]

[image]

[image]

[image]

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | v4.3.0 Full  |

marekpow, 2024-01-13:
And screenshot from ifconfig

[image]

brocaar, 2024-01-17:
Have you tried to connect to the wifi access-point that the GW OS exposes by default? As the default IP is 192.168.0.1, you might want to disable the AP mode: https://www.chirpstack.io/docs/chirpstack-gateway-os/getting-started.html#wi-fi.

Maybe we should change the default to a different less common IP? Anyway, disabling the wifi AP should resolve the issue.

marekpow, 2024-01-17:
Yes, it's the right direction to change the default IP to a less common IP. 
I edited /etc/config/network, set another address instead of 192.168.0.1, and set this IP address to reserved addresses on the router. It works now. 
CS OS 4.2 started on the RPi 3 (the web interface to the Open-Wrt works now).
I didn't do the same on the RPi 4 board yet.
For now, I'm trying to install RAK-5164 firmware on the RPi 4 board, but I don't know how to configure this gateway. I don't know a port to the web interface.

brocaar, 2024-01-31:
> Yes, it's the right direction to change the default IP to a less common IP.

Maybe it is better to just document that the IP might conflict. The `ChirpStackAP-xxxxx` is intended to bootstrap the network configuration, not to provide a permanent network interface. If the `192.168.0.1` is conflicting, then I think all it takes is temporarily disable this interface (e.g. unplug ethernet), configure the Raspberry Pi to connect to the Wifi (which will disable the AP mode), then reconnect to ethernet.

I have been looking at a less common IP, but it seems there is always someone using it.

brocaar, 2024-01-31:
@marekpow please see https://github.com/chirpstack/chirpstack-docs/commit/ffe13c99a5a7a417cd9cced8ac38a7204896d8b2

marekpow, 2024-01-31:
OK.

brocaar, 2024-03-28:
I'm closing this issue as I believe the above commit did clarify the issue you were experiencing (IP address conflict).