chirpstack/chirpstack-gateway-os #114: GPS not showing on gateway

dolence, 2024-06-21:
- [x ] The issue is present in the latest release.
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

Gateway doesn't show actual location on the web UI. I'm using an Elecrow LR1302 module and hat which is the same as the Seeed WM1302.

Firmware Version: ChirpStack Gateway OS 4.4.0 Full / LuCI openwrt-23.05 branch git-24.135.44542-f1ec9c2
Model: Raspberry Pi 3 Model B Rev 1.2

[image]

[image]

I can see GPS data on /dev/ttyAMA0. I suspect that the gps device path may be wrong on configuration files, but I don't know how to change this on Gateway OS since it's running on OpenWRT.

brocaar, 2024-06-24:
Are you sure your board has an UBLOX GNSS module? Because that is the only GNSS module that is supported by the Semtech HAL.

dolence, 2024-06-24:
It's a L76K from Quectel. It's the same used on Waveshare and SEED hats.

> Are you sure your board has an UBLOX GNSS module? Because that is the only GNSS module that is supported by the Semtech HAL.

Mohanbandaru16, 2024-08-10:
I'm using  L76K from Quectel. Any alternative fix or something to make it work?