chirpstack/chirpstack-gateway-os #115: 443 Forbidden web interface access on wifi Client mode

TregliaRaphael, 2024-06-21:
- [ x]  The issue is present in the latest release.
- [ x]  I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?
Hi, i own a 2287 spi rak module + pi3b. I had to swap the pin 17 -> 25 for reset.
Everythings works on AP mode.
The wifi client mode give me a 403 forbidden nginx/1.25.0. on blank screen error on the ip of my gateway

## What did you expect?
I expect to be able to connect the server and remove or fix the https restriction. It seems like certs are missing but i don't know how to set one simply with the current distrib

## Steps to reproduce this issue

Steps:

1.Set Client mode
2.Connect on the Gateway with ip given by router and not 192.168.0.1
3. 403 forbidden error nginx/1.25.0

## Your Environment
Pi3b + rak2287 spi
Master gateway os 4.4.0

Thank you for the help, i'll do my best to help you if needed

TregliaRaphael, 2024-06-21:
[image]

thomasesman, 2025-02-07:
I did encounter the same problem, but that was due to the fact that I tried to connect to the gateway from another subnet then the subnet that the gateway was in. You can solve this by adding the originating subnet to /etc/nginx/restrict_locally. Steps:
1. Login to the gateway using SSH
2. Edit /etc/nginx/restrict_locally with you favorite editor
3. Add the IP range of the originating subnet in the same format as the other IP ranges in the file
4. Save and close the file
5. Restart nginx: sudo /etc/init.d/nginx restart

This made it work for me.