chirpstack/chirpstack-gateway-os #120: No gateway stats on UDP forwarder mode ...

AngeLinuX99, 2024-10-27:
I have installed the latest version of chirpstack-gateway-os (v4.5.4) on a raspberry pi with an ic880a board and it is sending data via UDP forwarder (UDP/1700) to my chirpstack server (v4) that I have in production and hosted in the cloud. It always appears as "Never seen" in the list of gateways in my chirpstack.

The device is receiving radio packets on 868mhz well as you can see in the following log:

```
Sun Oct 27 23:28:38 2024 user.info chirpstack-concentratord-sx1301[2195]: Publishing stats event, rx_received: 2, rx_received_ok: 2, tx_received: 0, tx_emitted: 0
Sun Oct 27 23:29:08 2024 user.info chirpstack-concentratord-sx1301[2195]: Publishing stats event, rx_received: 0, rx_received_ok: 0, tx_received: 0, tx_emitted: 0
Sun Oct 27 23:29:15 2024 user.info chirpstack-concentratord-sx1301[2195]: Frame received, uplink_id: 2662950561, count_us: 1299435988, freq: 867900000, bw: 125000, mod: LoRa, dr: SF7
Sun Oct 27 23:29:38 2024 user.info chirpstack-concentratord-sx1301[2195]: Publishing stats event, rx_received: 1, rx_received_ok: 1, tx_received: 0, tx_emitted: 0
Sun Oct 27 23:30:08 2024 user.info chirpstack-concentratord-sx1301[2195]: Publishing stats event, rx_received: 0, rx_received_ok: 0, tx_received: 0, tx_emitted: 0
Sun Oct 27 23:30:15 2024 user.info chirpstack-concentratord-sx1301[2195]: Frame received, uplink_id: 3046779523, count_us: 1359421996, freq: 867900000, bw: 125000, mod: LoRa, dr: SF7
Sun Oct 27 23:30:38 2024 user.info chirpstack-concentratord-sx1301[2195]: Publishing stats event, rx_received: 1, rx_received_ok: 1, tx_received: 0, tx_emitted: 0
Sun Oct 27 23:31:08 2024 user.info chirpstack-concentratord-sx1301[2195]: Publishing stats event, rx_received: 0, rx_received_ok: 0, tx_received: 0, tx_emitted: 0
Sun Oct 27 23:31:15 2024 user.info chirpstack-concentratord-sx1301[2195]: Frame received, uplink_id: 2626790121, count_us: 1419407475, freq: 867300000, bw: 125000, mod: LoRa, dr: SF7
Sun Oct 27 23:31:38 2024 user.info chirpstack-concentratord-sx1301[2195]: Publishing stats event, rx_received: 2, rx_received_ok: 1, tx_received: 0, tx_emitted: 0
```

However, I don't see any gateway stats data being sent at any time. After enabling the MQTT forwarder option (without specifying any server, default tcp://localhost:1883 since I don't use MQTT forwarder), several MQTT errors start to appear on logs and the stats event appears now on gateway logs... but it never reaches my chirpstack server because i only have UPD/1700 opened on it.

```
Sun Oct 27 23:28:08 2024 user.info chirpstack-mqtt-forwarder[3062]: Received gateway stats
Sun Oct 27 23:28:08 2024 user.info chirpstack-mqtt-forwarder[3062]: Sending gateway stats event, topic: eu868/gateway/b827ebfffe11e0b8/event/stats
Sun Oct 27 23:28:09 2024 user.err chirpstack-mqtt-forwarder[3062]: MQTT error, error: I/O: Connection refused (os error 111)
Sun Oct 27 23:28:10 2024 user.err chirpstack-mqtt-forwarder[3062]: MQTT error, error: I/O: Connection refused (os error 111)
Sun Oct 27 23:28:11 2024 user.err chirpstack-mqtt-forwarder[3062]: MQTT error, error: I/O: Connection refused (os error 111)
Sun Oct 27 23:28:13 2024 user.err chirpstack-mqtt-forwarder[3062]: MQTT error, error: I/O: Connection refused (os error 111)
Sun Oct 27 23:28:14 2024 user.err chirpstack-mqtt-forwarder[3062]: MQTT error, error: I/O: Connection refused (os error 111)
Sun Oct 27 23:28:15 2024 user.err chirpstack-mqtt-forwarder[3062]: MQTT error, error: I/O: Connection refused (os error 111)
Sun Oct 27 23:28:15 2024 user.info chirpstack-concentratord-sx1301[2195]: Frame received, uplink_id: 4085891826, count_us: 1239451067, freq: 868300000, bw: 125000, mod: LoRa, dr: SF7
Sun Oct 27 23:28:15 2024 user.info chirpstack-mqtt-forwarder[3062]: Received uplink frame, uplink_id: 4085891826
Sun Oct 27 23:28:15 2024 user.info chirpstack-mqtt-forwarder[3062]: Sending uplink event, uplink_id: 4085891826, topic: eu868/gateway/b827ebfffe11e0b8/event/up
Sun Oct 27 23:28:16 2024 user.err chirpstack-mqtt-forwarder[3062]: MQTT error, error: I/O: Connection refused (os error 111)
```

Is it necessary to enable and setup MQTT forwarder on my chirpstack-gateway-os (v4.5.4) gateway and enable port 1883 on my chirpstack server that I have in the cloud to receive the stats events from my chirpstack-gateway-os (v4.5.4) ?? I have other gateway models from other manufacturers (Dragino, RAK, Milesight, SenseCAP) and only using UDP/1700 packet forwarder do they receive data and gateway status and I see the online status of the connected gateways.

Do somebody review this issue ?

Thanks

brocaar, 2024-10-28:
Please also share the logs of the `chirpstack-udp-forwarder`. Also make sure it is enabled (see the Enabled checkbox in the UI).

AngeLinuX99, 2024-10-28:
Hi brocaar, thanks for your quick reply

Service looks running, but no logs at all...
```

root@chirpstack-11e0b8:~# service chirpstack-udp-forwarder status
running
root@chirpstack-11e0b8:~# logread -f -l 1000 -e chirpstack-udp-forward
```
Also,no service found using ps 

```
root@chirpstack-11e0b8:~# ps w|grep chirpstack-udp-forwarder
 2915 root      1096 S    grep chirpstack-udp-forwarder
root@chirpstack-11e0b8:~# 

```

brocaar, 2024-10-28:
What is in your `/etc/config/chirpstack-udp-forwarder` logs?

AngeLinuX99, 2024-10-28:
config global
        option enabled '1'

config server
        option server '192.168.20.170:1700'

AngeLinuX99, 2024-10-28:
192.168.20.170:1700 is a machine that i have running netcat listener on udp/1700. Nothing arrives

brocaar, 2024-10-28:
I think with the refactor to multi-slot config, some config was not correctly ported. If you add this at the end of `etc/config/chirpstack-udp-forwarder`, the issue should be solved:

```
config concentratord
    option event_url 'ipc:///tmp/concentratord_event'
    option command_url 'ipc:///tmp/concentratord_command'
```

closed 2024-10-28 by brocaar

AngeLinuX99, 2024-10-28:
Issue solved, thanks brocaar !!!