chirpstack/chirpstack-gateway-os #122: Mesh gateway missing from bcm2708 build target

Ambrotos, 2025-02-11:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?
Attempting to navigate to the Mesh page in the LuCI GUI of the latest base image of Chirpstack Gateway OS results in an RPC call error saying resource not found. I believe this issue is specific to the bcm2708 build target.

## What did you expect?
The mesh page should either be included, or the option to navigate to it should be removed if it is not supported. I believe this issue is simply because the makefile for the bcm2708 build target is missing the line:
```
CONFIG_PACKAGE_chirpstack-gateway-mesh=y
```
I have cloned the repo, used make menuconfig to add the mesh gateway package to the bcm2708 build target, and compiled a local build. I have it installed on my raspberry pi 1 and everything seems to be running properly. Was this exclusion of the mesh gateway package from the bcm2708 build target accidental, or is there a specific reason gateway mesh functionality isn't supported on this build target?

## Steps to reproduce this issue

Steps:

1. Download, install, and boot the latest build of gateway OS for the bcm2708 target.
2. Attempt to navigate to the Mesh configuration tab.
3.
4.

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | v4.6.2  |

brocaar, 2025-02-12:
This was not intended. The above commit fixes it. Thanks for reporting 👍

closed 2025-02-12 by brocaar

Ambrotos, 2025-02-12:
Much obliged! Are you able to give an approx timeline until the next planned release? Until then I can continue to run my locally-built image, I'm just curious since there's a thread about this in the Chirpstack support forum and it would be nice to close the loop.

brocaar, 2025-02-14:
I expect pretty soon, 1 - 2 weeks. There are also some Gateway Mesh changes that I would like to include.

commit 5ca68bab (2025-02-12): rpi: Fix missing chirpstack-gateway-mesh in mbc2708.

Fixes #122.