chirpstack/chirpstack-gateway-os #127: Wifi firmware for Zero2 is missing

mschorer, 2025-04-13:
- [x ] The issue is present in the latest release.
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?

Wrote chirpstack-gateway-os-4.7.0-full-bcm27xx-bcm2709-rpi-2-squashfs-factory.img.gz to a SD card
Boot looks good - except kernel can't load the wifi firmware -> no networking

## What did you expect?

Firmware should load - networking be active.

## Steps to reproduce this issue

Steps:

1. write chirpstack-gateway-os-4.7.0-full-bcm27xx-bcm2709-rpi-2-squashfs-factory.img.gz to a micro sd
2. boot raspberry zero2 W
3. the failure(s) to find find the firmware with either of the 2 tried methods are at the end of the console output

## Could you share your log output?

currently no log - as no network.

## Your Environment

| Component             | Version |
| ----------------------| ------- |
| ChirpStack Gateway OS | v4.7.0  |

brocaar, 2025-04-14:
Could you confirm if the same issue persists when flashing a vanilla OpenWrt image for the same target?

brocaar, 2025-04-15:
> Openwrt 24.10.1 doesn't work either (openwrt-24.10.1-bcm27xx-bcm2709-rpi-2-ext4-factory.img.gz).

In that case, it would be better to report the issue here as this issue is not related to the ChirpStack Gateway OS, but affect all OpenWrt users:
https://github.com/openwrt/openwrt