chirpstack/chirpstack-gateway-os #133: gateway-os 4.7.1: openwrt 10.01.0 base kmod package src missing

g6094199, 2025-07-08:
- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-os/issues) of this repository and believe that this is not a duplicate.

## What happened?
cant install zram-swap

## Steps to reproduce this issue

Steps:

1. opkag update
2. opkg install zram-swap
3. 

> Unknown package 'zram-swap'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-zram for zram-swap
 * pkg_hash_fetch_best_installation_candidate: Packages for zram-swap found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package zram-swap.

## Solution

kmod src is  missing from opkg sources:
`src/gz openwrt_kmods https://downloads.openwrt.org/releases/24.10.1/targets/bcm27xx/bcm2708/kmods/6.6.86-1-????????????????`

## Your Environment

> root@PHI-Test-GW:/etc/opkg# cat /etc/openwrt_release 
> DISTRIB_ID='ChirpStack Gateway OS'
> DISTRIB_RELEASE='4.7.1'
> DISTRIB_REVISION='r28597-0425664679'
> DISTRIB_TARGET='bcm27xx/bcm2708'
> DISTRIB_ARCH='arm_arm1176jzf-s_vfp'
> DISTRIB_DESCRIPTION='ChirpStack Gateway OS 4.7.1 Full'
> DISTRIB_TAINTS='no-all'
>

brocaar, 2025-08-27:
I'm sorry, I might have been too fast in my reaction. This is not going to work as the kernel configuration of the ChirpStack Gateway OS is not exactly the same as the "official" OpenWrt image. To install additional kernel-modules, you need to build the ChirpStack Gateway OS from source.

See last bullet on page: https://openwrt.org/docs/guide-developer/toolchain/beginners-build-guide

g6094199, 2025-08-27:
hmmm. ok. i see. 
i built openwrt from time to time from source.
it should work as long as the openwrt standard kernel swithces are enabled this feed should work. no matter how man  additional switches are set.
maybe we can try it?